EFeru/hoverboard-firmware-hack-FOC #2: Misc. questions

gnbl, 2019-10-22:
Hi,
I finally managed to get hold of an "old" STM-based hoverboard and awed by your video, I want to hook up my ST-Link and get the wheels spinning -- but I have a few questions and maybe you could take some time off improving the code and add a few pointers to the readme... :-)
- difference/modifications to https://github.com/NiklasFauth/hoverboard-firmware-hack ? (replace, extend control with FOC, at what "interfaces" exactly? usable as drop-in replacement?)
- what is the default configuration in the repository?
- CPU load?
- hardware requirements (current sensing, boards supported)
- getting started: Choosing speed/steering command input (ADC, Nunchuck, Serial) and what behaviour to expect in conjunction with "control modes" (voltage/speed/torque) and "control methods"  (trapezoidal/FOC/..)- Can these be switched on the fly? Use cases?
- what would need to be done to add a balancing control loop (or two) to restore hoverboard functionality?
- motor model in Matlab; does this need to be touched for similar motors and when, what is required to do so (what's a webview?), are there any tuning hints for the parameters?
- what is your motor (model, specs, any information really), how many have you tested, etc. Also, I've read that motor temperature can have quite an effect and "throw off" the model - have you tested this as well or is this negligible?

EmanuelFeru, 2019-10-22:
Quite intensive questions :) Please find below my answers:

-----------------
> * difference/modifications to https://github.com/NiklasFauth/hoverboard-firmware-hack ? (replace, extend control with FOC, at what "interfaces" exactly? usable as drop-in replacement?)

► It is actually a replacement of the original firmware for people that want high motor performance (lower noise, improved efficiency and improved safety via better diagnostics and current protection).
► Regarding interfaces, to the to the user are the same, the controller uses now the phase current measurements.
► Usable as drop-in replacement: Yes

-----------------
> * what is the default configuration in the repository?

► User input: ADC potentiometer
►  Mode: Voltage mode
► Control type: FOC
► Diagnostics: Enabled
► Field weakening: Disabled
► Max motor current: 15 A/motor
► Max speed: 800 rpm
All this can be found in [config.h](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Inc/config.h)

-----------------
> * CPU load?

► CPU load FOC:
- 85.4% 53ms (VOLTAGE mode) 
- 88.7% 55ms (SPEED mode)
- 90.3% 56ms (TORQUE mode)
Measurement method: by the DMA execution time, both motors spinning. Available time ~62 ms (16 kHz).
► Memory (with current commit):
DATA:    [=         ]   6.7% (used 3276 bytes from 49152 bytes)
PROGRAM: [=         ]   7.1% (used 18656 bytes from 262144 bytes)

-----------------
> * hardware requirements (current sensing, boards supported)

► Current sensing: yes (mandatory for FOC)
► Boards supported: single boards for now (same as for original firmware)

-----------------
> * getting started: Choosing speed/steering command input (ADC, Nunchuck, Serial) and what behaviour to expect in conjunction with "control modes" (voltage/speed/torque) and "control methods"  (trapezoidal/FOC/..)- Can these be switched on the fly? Use cases?

► Command input: same availability as with original firmware. Recommended to start with: ADC potentiometer
► Behavior of "control modes": described in ReadMe, switchable on the fly: yes (the code will aromatically ramp down the motors to stop and then perform the switch).
► "control methods" (trapezoidal/FOC): yes, switchable on the fly.

-----------------
> * what would need to be done to add a balancing control loop (or two) to restore hoverboard functionality?

► Shall be possible, please consult [btsimon repo](https://github.com/bipropellant/bipropellant-hoverboard-firmware)

-----------------
> * motor model in Matlab; does this need to be touched for similar motors and when, what is required to do so (what's a webview?), are there any tuning hints for the parameters?

► Motor model in Matlab: this is a functional model (not validated), meaning it is used to develop functionality and detect issues at an early stage without breaking the hardware. It is supposed to stay this way. Thus, no need to change it for a similar motor.
► Fine tuning is performed on the real hardware/motor
► Hints for parameters tuning: basic motor control knowledge and then you are good to go. On need, I can give specific hints if requested.
► Webview: This is just a multitude of html files which allows to visualize the model without matlab. Give it a try by running the file webview.html from [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/tree/master/01_Matlab/BLDC_controller_ert_rtw/html/webview)

-----------------
> * what is your motor (model, specs, any information really), how many have you tested, etc. Also, I've read that motor temperature can have quite an effect and "throw off" the model - have you tested this as well or is this negligible?

► See previous answer. For exact motor model parameters see the Simulink model. I put the parameters that I found online for such hoverboard motors, regarding poles, phase resistance and inductance. These are not exactly the same as in the real hardware but it is not a problem, it is a functional model and the final tuning is done on the real hardware.
► Temperature until now seems negligible, needs to be checked further.

gnbl, 2019-10-22:
Awesome, thanks. I might be able to try it tomorrow.

gnbl, 2019-10-23:
Hello again, 
I've tried to have a look at the webview, however, it does not work for me locally for security reasons (with up-to-date Firefox) if not coming from an actual web server ("CORS request not http").

Since I have installed Python, I could start a simple web server in the webview directory like so:
`python -m http.server 8000` (https://docs.python.org/3/library/http.server.html) and direct the browser to `http://localhost:8000/webview.html`.

This is what it looks like (feel free to use the file):

[image]

You could see if this can be somehow "mapped" to GitHub user pages (https://guides.github.com/features/pages/) but it seems that requires a separate repository (submodule?).

EmanuelFeru, 2019-10-23:
True, by default some browsers (e.g., Chrome) does not allow javascript  to interact with local file systems due to security reasons. They give indications how to do it [here] (http://www.chrome-allow-file-access-from-file.com/)

To make it easy, I suggest to use the following browsers: Internet Explorer or Microsoft Edge. For both worked.

closed 2019-10-24 by EmanuelFeru