EFeru/hoverboard-firmware-hack-FOC #3: ADC calibration not possible as raw values are not in debug output

gnbl, 2019-10-23:
Problem: cannot calibrate as per https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Inc/config.h#L98 because the required ADC raw values for speed (ADC2) and steering (ADC1) are not in debug output due to incomplete / commented code from 
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/main.c#L336

It probably should look like:
```
// ####### DEBUG SERIAL OUT #######
      #ifdef CONTROL_ADC
      setScopeChannel(0, (int)adc_buffer.l_tx2);        // 1: ADC1
      setScopeChannel(1, (int)adc_buffer.l_rx2);        // 2: ADC2
      #else
      setScopeChannel(0, (int16_t)speedR);                    // 1: output command: [-1000, 1000]
      setScopeChannel(1, (int16_t)speedL);                    // 2: output command: [-1000, 1000]
      #endif
```
or just get extended to output all these values.

(the wheels are spinning, albeit with some noise from the control board)

gnbl, 2019-10-23:
Also: Debug output rate depends on motor speed (ADC1 = 300, ADC2 = 0 results in visible output rate of  a few Hertz)?

EmanuelFeru, 2019-10-24:
- ADC calibration: ok, I will put that back by un-commenting
- what do you mean by "the wheels are spinning, albeit with some noise from the control board"? what kind of noise?
>  Debug output rate depends on motor speed (ADC1 = 300, ADC2 = 0 results in visible output rate of a few Hertz)?

Please explain in more detail what "Debug output rate"? Serial? And what exactly do you see is happening.

EmanuelFeru, 2019-10-24:
> * "the wheels are spinning" can be disregarded. It was meant to report some success running your firmware. The noise (high frequency buzzing/whirring) I hear is much more audible than what I could hear in your video, so either your camera did not pick that up or it is related to my hardware.

► ok, ok :) Regarding noise, it can be a lot of things, maybe camera difference, room, placement of the hoverboard, etc.. The level of noise also depends on the speed you are operating. With the video I only wanted to demonstrate that there is a big difference between Commutation control and FOC.

> * Yes, I meant the serial debug output from https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/main.c#L349. It appears to be proportional to motor speed, e.g.
>   SpeedR=0, SpeedL=0 -> debug output every 150 ms.
>   SpeedR=-52, SpeedL=-52 -> debug output every 60 ms.

► This is normal, because of the way **inactivity_timeout_counter** is incremented. When the Input is above 50 it is displayed more often, because there are things changing. like motors spinning When Input is low (motor is stopped), it is done more rarely because there is not much going on. So, it not an issue, it is intended behavior. See these lines:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1cc8b3475fc2929dea1685a7dc824458ae5e8e9e/Src/main.c#L330
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1cc8b3475fc2929dea1685a7dc824458ae5e8e9e/Src/main.c#L385-L390

gnbl, 2019-10-24:
OK, thanks for the feedback!
I will switch the commutation mode temporarily to be able to tell the difference.

EmanuelFeru, 2019-10-27:
Implemented here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1376793710294773692ef9a4df178f2e6f0b5665/Src/main.c#L351-L354

closed 2019-10-27 by EmanuelFeru