EFeru/hoverboard-firmware-hack-FOC #4: "driving behavior" - reverse (backward) currently not possible?

gnbl, 2019-10-23:
What changes are required to allow "reversing" of the speed? Should this be done at the CONTROL_xxx level (main.c) or would the introduction of e.g. SPEED_OFFSET and STEER_OFFSET be preferable?

- https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/main.c#L268
- https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Inc/config.h#L144

Btw, 
```
#define INVERT_R_DIRECTION
#define INVERT_L_DIRECTION
```
seem both to be defined by default, but there is no explanation as to why this is required in addition to SPEED_COEFFICIENT.

gnbl, 2019-10-24:
I think reversing should be possible at runtime by input, unless you want the scope of this firmware to be limited to one direction (two-wheelers like scooters etc.).

EmanuelFeru, 2019-10-24:
> I think reversing should be possible at runtime by input, unless you want the scope of this firmware to be limited to one direction (two-wheelers like scooters etc.).

Of course, indeed, you can always reverse the direction by input. The parameters I was referring is for establishing the behavior of the input.. more exactly sign of the input in connection with sign of the motor speed.

gnbl, 2019-10-24:
That's what this does:

https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1cc8b3475fc2929dea1685a7dc824458ae5e8e9e/Src/main.c#L287-L289

EmanuelFeru, 2019-10-24:
yes, you can use that. Although the input resolution would be half, because we go from [0, 1000] -> [-1000, 1000] (using integer data type).
If I have time, I will think of something, immediately after the ADC is read to not loose resolution.

gnbl, 2019-10-24:
How about this?
```
      cmd1 = (CLAMP(adc_buffer.l_tx2 - ADC1_MIN, 0, ADC1_MAX) * 2000 / ADC1_MAX) - 1000;  // ADC1 steer
      cmd2 = (CLAMP(adc_buffer.l_rx2 - ADC2_MIN, 0, ADC2_MAX) * 2000 / ADC2_MAX) - 1000;  // ADC2 speed
```

EmanuelFeru, 2019-10-27:
Implemented here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1376793710294773692ef9a4df178f2e6f0b5665/Src/main.c#L270-L282

closed 2019-10-27 by EmanuelFeru