EFeru/hoverboard-firmware-hack-FOC #5: consoleLog() not configured

gnbl, 2019-10-24:
This should use the #defines - it's defaulting to the wrong peripheral/channel:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1cc8b3475fc2929dea1685a7dc824458ae5e8e9e/Src/comms.c#L64

gnbl, 2019-10-24:
Mhh, but even if this is set to huart3, I can't seem to output anything, e.g.

https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/main.c#L146
```
  #if defined(DEBUG_SERIAL_USART2) || defined(DEBUG_SERIAL_USART3)
    UART_Init();
    consoleLog("Hello world!");
    ...
```

I'll look into it further in a couple of hours.

gnbl, 2019-10-24:
I've dabbled with this a bit and got some output - however, multiple consecutive calls are not working, which likely is caused by how DMA works. I have a feeling this might be improved using a ring buffer.

I've basically duplicated the code to use the UART. Also, I don't think memset is necessary since no data beyond the string length is sent. And strlen can be eliminated by using sprintf's return value. sprintf writes a terminating \0 anyway (run online: https://repl.it/repls/DarkShamefulGoal).

- https://en.cppreference.com/w/c/io/fprintf

```

#include "comms.h"
#include <stdio.h>
#include <string.h>
#include "config.h"
#include "defines.h"
#include "setup.h"
#include "stm32f1xx_hal.h"

#ifdef DEBUG_SERIAL_USART3
#define UART_DMA_CHANNEL DMA1_Channel2
UART_HandleTypeDef huart3;
#endif

#ifdef DEBUG_SERIAL_USART2
#define UART_DMA_CHANNEL DMA1_Channel7
UART_HandleTypeDef huart2;
#endif

static volatile uint8_t uart_buf[100];
static volatile int16_t ch_buf[10];

void setScopeChannel(uint8_t ch, int16_t val) { ch_buf[ch] = val; }

void consoleScope(void) {
#if defined DEBUG_SERIAL_SERVOTERM && \
    (defined DEBUG_SERIAL_USART2 || defined DEBUG_SERIAL_USART3)
  uart_buf[0] = 0xff;
  uart_buf[1] = CLAMP(ch_buf[0] + 127, 0, 255);
  uart_buf[2] = CLAMP(ch_buf[1] + 127, 0, 255);
  uart_buf[3] = CLAMP(ch_buf[2] + 127, 0, 255);
  uart_buf[4] = CLAMP(ch_buf[3] + 127, 0, 255);
  uart_buf[5] = CLAMP(ch_buf[4] + 127, 0, 255);
  uart_buf[6] = CLAMP(ch_buf[5] + 127, 0, 255);
  uart_buf[7] = CLAMP(ch_buf[6] + 127, 0, 255);
  uart_buf[8] = CLAMP(ch_buf[7] + 127, 0, 255);
  uart_buf[9] = '\n';

  if (UART_DMA_CHANNEL->CNDTR == 0) {
    UART_DMA_CHANNEL->CCR &= ~DMA_CCR_EN;
    UART_DMA_CHANNEL->CNDTR = 10;
    UART_DMA_CHANNEL->CMAR = (uint32_t)uart_buf;
    UART_DMA_CHANNEL->CCR |= DMA_CCR_EN;
  }
#endif

#if defined DEBUG_SERIAL_ASCII && \
    (defined DEBUG_SERIAL_USART2 || defined DEBUG_SERIAL_USART3)
  // memset((void *)(uintptr_t)uart_buf, 0, sizeof(uart_buf));
  int rv = sprintf((char *)(uintptr_t)uart_buf,
          "1:%i 2:%i 3:%i 4:%i 5:%i 6:%i 7:%i 8:%i 9:%i A:%i\r\n", ch_buf[0], ch_buf[1],
          ch_buf[2], ch_buf[3], ch_buf[4], ch_buf[5], ch_buf[6], ch_buf[7], ch_buf[8], ch_buf[9]);

  if(rv<0){
    return;
  }

  if (UART_DMA_CHANNEL->CNDTR == 0) {
    UART_DMA_CHANNEL->CCR &= ~DMA_CCR_EN;
    // UART_DMA_CHANNEL->CNDTR = strlen((char *)(uintptr_t)uart_buf);
    UART_DMA_CHANNEL->CNDTR = rv;
    UART_DMA_CHANNEL->CMAR = (uint32_t)uart_buf;
    UART_DMA_CHANNEL->CCR |= DMA_CCR_EN;
  }
#endif
}

void consoleLog(char *message) {

/*
  #ifdef DEBUG_SERIAL_USART3
HAL_UART_Transmit_DMA(&huart3, (uint8_t *)message, (uint16_t)strlen(message));
#endif

#ifdef DEBUG_SERIAL_USART2
  HAL_UART_Transmit_DMA(&huart2, (uint8_t *)message, (uint16_t)strlen(message));
#endif
*/

#if defined DEBUG_SERIAL_ASCII && \
    (defined DEBUG_SERIAL_USART2 || defined DEBUG_SERIAL_USART3)

  if (UART_DMA_CHANNEL->CNDTR == 0) {
    UART_DMA_CHANNEL->CCR &= ~DMA_CCR_EN;
    UART_DMA_CHANNEL->CNDTR = strlen((char *)(uintptr_t)message);
    UART_DMA_CHANNEL->CMAR = (uint32_t)message;
    UART_DMA_CHANNEL->CCR |= DMA_CCR_EN;
  }
#endif
}

```

EmanuelFeru, 2019-10-27:
Not very clear what needs to be done for this issue...

gnbl, 2019-11-01:
> Not very clear what needs to be done for this issue...

Does it work for you? Do you need additional info?

EmanuelFeru, 2019-11-03:
@gnbl do you want to control the board via serial USART? So, that the board receives command signals from serial rather than ADC? Or more generic question how do you want control the board?
Because I did not test the functionality of control via serial, so there might be issues.

gnbl, 2019-11-03:
As far as I understand, `consoleLog()` has nothing to do with control (I have not checked in that context), but rather debugging - which I was not able to use it for. 
If you can reproduce the problem, you could either fix it somehow or disable/remove the function altogether.

With the CPU usage you mentioned it might be better to use a separate controller in order to not complicate things and also get more IO easily. In that case, I would not need the ADCs but one serial interface (UART probably being the easiest to debug).

EmanuelFeru, 2019-11-07:
@gnbl I plan to work on this in the future. I will try to implement full bidirectional control on USART3. So, that the board can be controlled by sending Serial commands and also receive feedback from the board to know what is it doing.

For now, if you want to debug  you just need to read the Serial data that is sent on USART3 (short original sensor wire **without the Red wire because it carries 15V!!**). An then using a Serial Monitor you should be able to receive data in the following form:
```
1:0 2:0 3:0 4:0 5:1800 6:3856 7:1800 8:325
```
meaning:
```
1:ADC1 2:ADC2 3:speedR 4:speedL 5:ADC_batVoltage 6:batVoltage 7:ADC_boardTemp 8:boardTemp
```
In the example above you have: speedR  = 0, speedL = 0, batVoltage  = 38.56 V, boardTemp = 32.5 degC

EmanuelFeru, 2019-11-08:
@gnbl indeed the consoleLog() is not working. I was trying to fix it. Did you manage to achieve something?
For me it just hangs, and the only way to shut-down the battery is by disconnecting the battery.

gnbl, 2019-11-09:
Hi Emanuel, I have not spent any more time on this after getting some debug output working the way I described [earlier](#issuecomment-546044028).

EmanuelFeru, 2019-11-09:
@gnbl I tried your approach and it works if it is not called more than once after each other. Because it needs some time to send otherwise the if condition is false and it will l not send. 
However, if it is not called immediately it works nicely. I will go with your approach for now.

EmanuelFeru, 2019-11-29:
Bug fixed.

closed 2019-11-29 by EmanuelFeru