EFeru/hoverboard-firmware-hack-FOC #6: Don't power off when releasing the power button after flashing

gnbl, 2019-10-24:
I've been pushing the power button when flashing and was annoyed that the board would power off when I released it. This change seems to fix that, though it probably would be better to handle/copy the flags directly after reset:
```
    // ####### POWEROFF BY POWER-BUTTON #######
    // unless the button is still being pressed after there was a software reset (programmer/debugger)
    if (HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN) ) {

      // disable motors
      enable = 0;

      // wait until button is released
      while (HAL_GPIO_ReadPin(BUTTON_PORT, BUTTON_PIN)) {}

      // don't power off immediately after software reset
      if( __HAL_RCC_GET_FLAG(RCC_FLAG_SFTRST) ) {
        // clear reset flags
        __HAL_RCC_CLEAR_RESET_FLAGS();
      } else {
        // release power latch
        poweroff();
      }
    }
```

gnbl, 2019-10-27:
Just for reference, "bipropellant" uses lots of additional logic for the power button (also mentioning STLink):
- https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/master/src/main.c#L121
- https://github.com/bipropellant/bipropellant-hoverboard-firmware/blob/master/src/main.c#L940

EmanuelFeru, 2019-10-27:
@gnbl I think in bipropellant, he detects if the button is pressed for longer time to enable, the calibration of the gyro, for leveling the hoverboard. This is only needed in hoverboard mode.
For now, I will implement your suggestion.

EmanuelFeru, 2019-10-27:
Implemented here
 https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1376793710294773692ef9a4df178f2e6f0b5665/Src/main.c#L365-L373

closed 2019-10-27 by EmanuelFeru