EFeru/hoverboard-firmware-hack-FOC #7: Single motor operation

Blimpyway, 2019-10-25:
Hi,

I need to control only one motor e.g. left one. 
If second motor (right)  is disconnected, apart from diagnosing it as faulty, does the left side motor works as usual?

Thanks and I apologise for not testing it myself, I have no experience with reflashing the main board just want to know if I can start to learn it or use a different controller.

EmanuelFeru, 2019-10-26:
Hi @Blimpyway ,

Yes, controlling one motor is possible and actually very easy to do by commenting out one line in the code.
e.g., if you want to control only the Left motor, then just **comment out the Right motor** here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/1cc8b3475fc2929dea1685a7dc824458ae5e8e9e/Src/bldc.c#L209

The diagnostics will not trigger because the errorCode for the Right motor will be 0 (no error) since the function is not called.

Blimpyway, 2019-10-27:
Thanks a lot!

closed 2019-10-27 by EmanuelFeru