EFeru/hoverboard-firmware-hack-FOC #13: Wow, super cool variants, one thing more please

abrarakram19, 2020-01-11:
I am hobbyist specially in field of electronics, ESCs and brushless motors. I have many HB main boards for experiments. I am very excited to see this project. My request is can you please add "sensor less bldc cotrol" variant? Thanks for your great open source FW.

pinguinpfleger, 2020-01-19:
There are hoverboards out there with unsensored motors???
What is bidirectional operation?

EmanuelFeru, 2020-01-25:
@abrarakram19  thanks for the nice words! . This is already bidirectional (spins in both directions).
Regarding the sensor-less, unfortunately I will not spend the effort on it  because my intention is not to use this board for other motors (at least for now).