EFeru/hoverboard-firmware-hack-FOC #21: USART Variant wont run

cohnhead66, 2020-02-13:
When I build the USART Variant and upload the firmware to the board i get continuous "harsh' beeps and no response from any serial commands.

All works well when I build the ADC Variant and run motors with potentiometers.  

What do the harsh beeps indicate?  I assume it means firmware isn't right.

Can you provide guidance on how to properly build the USART variant.  What lines to change in config.h

appreciate any help

EmanuelFeru, 2020-02-14:
In case you use USART variant, then you have to provide input commands as given in this [Arduino example](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/02_Arduino/hoverserial/hoverserial.ino) with Start frame and checksum. 
Otherwise, the board will go into a Serial Timeout (because of no proper command) and it will do low pitch beeping (this you already experienced).
Maybe at the first glance is annoying, but if you are driving this safety feature can save some injury by making the board not go away by itself.

cohnhead66, 2020-02-14:
Thank you,  I figured out that my serial connection was unreliable at the implemented BAUD rate.  Once I addressed that, USART Variant works as described.

EmanuelFeru, 2020-02-16:
@cohnhead66 What baud rate did you use?

cohnhead66, 2020-02-17:
Used baud rate 38400.  

I am concerned that serial only works when I send it 5v signal.  I tried to use a 5v to 3.3v logic level converter originally, but there was no communication.  I used the same logic level converter to establish a serial connection using the build from this repository  https://github.com/Combinacijus/hoverboard-firmware-hack and it worked at 115200 baud, so i know the logic level converter works as designed.  

I have also tried to send 3.3v signal directly from a 3.3v version of an arduino pro mini and could not establish communication.  

Have you had any issues using a 5v signal?  I have seen warnings on the original read me that this could damage the board.

EmanuelFeru, 2020-02-17:
For me it worked with an Arduino nano which operates at 3.3V .  See a short demo video here:
https://drive.google.com/open?id=1mUM-p7SE6gmyTH7zhDHy5DUyczXvmy5d

cohnhead66, 2020-02-17:
How did you modify Arduino nano to run at 3.3v?

EmanuelFeru, 2020-02-17:
Oh, I just saw that indeed Arduino nano pins are from [0,5V], I thought it was 3.3V, but no.
I did use the nano only for some time, when I developed the reliable UART communication and I didn't have issues. I am not sure what will happen on long term, maybe you are right.. I don't know, there are resistors on the mainboard to limit the current.. but I am not sure if it still a problem.

I don't know exactly when, but I will connect the original sideboard (which operates at 3.3V) to the mainboard. Maybe that will help to see the issue. If you find something meanwhile, let me know.

cohnhead66, 2020-02-18:
Everything appears to be working as designed on the UART end.  I was able to communicate at 3.3V using a logic level converter by correcting my wiring.  I havn't been able to communicate with my 3.3v pro mini board, but pretty sure that is on my end, and not on the hoverboard firmware side.

Will close this issue.

Thanks for your help.

closed 2020-02-18 by cohnhead66