EFeru/hoverboard-firmware-hack-FOC #22: Is there Hoverboard Mode?

tonvanhieu, 2020-02-13:
Hi. Can i combine this repo with your hoverboard-sideboard-hack to make my hoverboard works same as a normal hoverboard?
 
If it's possible, which VARIANT i should use?

Thank you very much.

EmanuelFeru, 2020-02-14:
I would be more than happy if you will try that out 👍  because I am also planning to add this feature in the future. Help will be very welcome.
In the mainboard branch we need to create VARIANT_HOVERBOARD (which is new, not yet created) and work there. It will be a special USART variant, where we enable both USARTs in full duplex-mode.
On the sideboards, we use also VARIANT_HOVERBOARD (already created,  however it might require some modification to match the mainboard VARIANT_HOVERBOARD).

If you have results let me know 👍

tonvanhieu, 2020-02-15:
Yeah, i will try my best. 
I have one more question please. Can we build that project by Keil C.
I've tried but always had troubles with the hal_timebased drivers
Thank you very much

EmanuelFeru, 2020-02-16:
For me that is fine.. what sideboard type do you have? By hearing HAL I guess you have the STM version.
My aim is to keep the GD version also aligned, because from what I see the GD version is more popular than STM.

tonvanhieu, 2020-02-16:
I have GD32 version.

Oh my stupid expression. " Can we build that project by Keil C" -----> i mean this repo, FOC firmware. Can you have anyway in order to build, flash and debug this repo in Keil?

I see in other repos, they say that we can flash STM32 program to GD32, is that true?

EmanuelFeru, 2020-02-16:
For the FOC repo I don't have a Keil project but is easy to set-up. I will prepare quickly one for you and commit. 
Regarding the flashing, indeed I heard as well that the GD for the mainboard is a clone of the STM, so that means it can be flashed with the same (I haven't tried myself though).
However, for the sideboards they are completely different, memory size, internal mapping, so it is not possible.

tonvanhieu, 2020-02-16:
Oh, Thank you very much. 
I am going to make Hoverboard mode by Kalman filter + fuzzy control. When it's done i will tell you.
Thank you again mate.

EmanuelFeru, 2020-02-16:
done. Check the Keil project [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/tree/master/MDK-ARM)
I created also the define for the VARIANT_HOVERBOARD to work on, check [config.h](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Inc/config.h#L325-L333)

tonvanhieu, 2020-02-17:
Thank you so much,
I appreciate your enthusiasm very much.
I will close this issue as well.

closed 2020-02-17 by tonvanhieu