EFeru/hoverboard-firmware-hack-FOC #24: 67v power possible?

gerkon56, 2020-02-20:
Hi. I want to use a 65volt battery . But the Board turns on and off immediately. Although everything works from 40v. Please tell me what to correct for work from 65 V. Thanks.

gerkon56, 2020-02-28:
Tried to configure bat adc. Changed values, more than 2000 to expose gives an error. Replaced the R54 resistor to fool the controller but still nothing comes out. The controller is turned on and immediately disabled. Although the sine project works. Who knows what else you can try to configure for the fok project from 67v.

EmanuelFeru, 2020-03-01:
I guess it was due to Filtering of the battery voltage in Fixed point.
Check my last commit, it should be fixed :)

gerkon56, 2020-03-01:
Thanks a lot. I still flashed the sine project. On 65 volt works. Under load, it is heated only to 50-60 degrees.

gerkon56, 2020-03-02:
Verified. Powered by 67v, does not turn off. But if you turn on the fok, then after the middle of the throttle stick, the wheels turn sharply and they turn in the opposite direction. Tried options with degrees fok. And adc level 2. Sometimes it helps, but it's not stable

EmanuelFeru, 2020-03-02:
That could be. FOC is using current measurement. So, it might be that some gains have been changed for the current ADC readings... or just simply the current readings are affected by the increased Voltage, and the system responds differently.. for sure the system behavior changed. 

For your case I recommend Sinusoidal Control type, to avoid strange behavior. That works quite well too, it just misses Torque Mode and Speed mode.

closed 2020-03-24 by EmanuelFeru

gerkon56, 2020-03-25:
Yes sinus is fine work. Thanks!

blubbi321, 2020-06-15:
I was considering to upgrade my power supply to ~48V .. so given that @gerkon56 didnt have issues on the hardware side Id assume that the upgrade wont break anything on the board?

gerkon56, 2020-06-15:
Everything will be fine. I use 65v and fok without weakening the field. The only thing I replaced on the Board native dc-dc to dc Board with aliexpress. The native dc-dc transistor was heated at 65v.