EFeru/hoverboard-firmware-hack-FOC #25: Freewheel mode?

cohnhead66, 2020-02-21:
There are times in my application where i would like to push my cart manually.  Any way to put the board in a mode that allows the wheels to spin freely with no resistance.  Currently I have to turn the board off completely.  I would rather switch to a freewheel mode from my remote.

cohnhead66, 2020-02-25:
@EmanuelFeru  Is there a USART protocol for any commands other than the setting the speed and steer?  If not, can you add protocol for:
-- Enable/Disable  of motors
-- Change of Mode
-- Buzzer control

Thanks

pinguinpfleger, 2020-02-26:
not yet:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/main.c#L107

the switch for the motors is called [enable](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/main.c#L175)

Buzzer Control is in the making, I hope I can contribute it soon

cohnhead66, 2020-02-27:
Thanks @pinguinpfleger.

for the enable switch, I don't see any method of setting the enable flag through a USART command.  It looks like only a SerialCommand struct can be sent to the firmware.

Sorry if I am missing something easy

pinguinpfleger, 2020-02-27:
right, you have to extend the struct and implement the functionality

cohnhead66, 2020-02-27:
I was afraid of that.  Afraid that might be a little over my skill level.  Let me take a look, might need some help.

EmanuelFeru, 2020-03-01:
I just added a commit in which you can use one sideboard as additional input to change modes, enable disable Field weakening, check my last commit :)

For your thing, you can put the board in TORQUE mode if you want to push it, then it will move easily.

cohnhead66, 2020-03-02:
Thanks @EmanuelFeru, I will look at this and try figure out how to extend the USART struct to enable this feature

EmanuelFeru, 2020-03-02:
It is already done. I added support for sideboard, including UART communication, push buttons, check the VARIANT_HOVERCAR to see how is done.

cohnhead66, 2020-03-03:
@EmanuelFeru   I think I understand.  I can send a SerialSideboard struct matching:
    typedef struct{
      uint16_t	start;
      int16_t  	roll;
      int16_t  	pitch;
      int16_t  	yaw;
      uint16_t  sensors;
      uint16_t 	checksum;
    } SerialSideboard;

to the USART that I am not using for steer and speed control.  

I just need to know the value as the sensor parameter. Not sure what that looks like when I just want to send it from my serial port.

I see you set this on your sideboard via this line:
Sideboard.sensors		= (uint16_t)(sensor1 | (sensor2 << 1) | (mpuStatus << 2));

Thanks again for your help.

EmanuelFeru, 2020-03-04:
If you send it in this form
```
Sideboard.sensors = (uint16_t)(sensor1 | (sensor2 << 1) | (mpuStatus << 2));
```
it should be fine. This shows: 1st bit = sensor1, 2nd bit = sensor2, 3rd bit = mpuStatus,
So if you send for example `0x0005` that means sensor1 = 1, sensor2 = 0, mpuStatus = 1.

cohnhead66, 2020-03-10:
Thank you.

I ended up extending the command struct for USART to include ctrlMode.  

  typedef struct{
      uint16_t  start; 
      int16_t   steer;
      int16_t   speed;
      int16_t   ctrlMode;
      uint16_t  checksum;    
    } Serialcommand;

and then copy ctrlMode to ctrlModReqRaw in the readCommand function.  Seems to work for me.  

I am now able to switch between TORQUE mode and SPEED.

closed 2020-03-10 by cohnhead66

pav18, 2020-04-21:
Is it possible to switch MODE without sideboard?

RoboDurden, 2022-06-23:
I had the same problem with automating the rotary crank of my log band saw ( https://youtu.be/PGM7NyRvg-c )

So i added
`//#define ENABLE_TIMEOUT        3       // Seconds of not driving will enable=0 for power saving and freewheeling
`
to the `DEFAULT SETTINGS `in https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/dbc485bbf480f91cf15851133c9cda170b30e965/Inc/config.h#L177

In main.c:
```
    #ifdef ENABLE_TIMEOUT	// robo
      enable2 = inactivity_timeout_counter <= (ENABLE_TIMEOUT * 1000) / (DELAY_IN_MAIN_LOOP + 1);
    #endif
```
after https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/dbc485bbf480f91cf15851133c9cda170b30e965/Src/main.c#L584

with 
`extern uint8_t enable2;                  //robo global variable for additional ENABLE_TIMEOUT motor enable
`
after https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/dbc485bbf480f91cf15851133c9cda170b30e965/Src/main.c#L83

and in bldc.c:
`uint8_t        enable2       = 1;        // global variable for additional ENABLE_TIMEOUT 
`
and
`  if(ABS(curL_DC) > curDC_max || enable == 0 || enable2 == 0) {
`
and
`  if(ABS(curR_DC)  > curDC_max || enable == 0 || enable2 == 0) {
`
and
`  enableFin = enable && enable2 && !rtY_Left.z_errCode && !rtY_Right.z_errCode;		// robo
`

I chose a new variable `enable2 `to not overwrite the `enable `variable whichs is also changed in the `VARIANT_TRANSPOTTER `lines, etc.
Maybe someone would like to commit it..