EFeru/hoverboard-firmware-hack-FOC #26: support of motor in delta variant

Rudi84, 2020-02-26:
@EmanuelFeru : Thanks for the nice working FOC Hack. I will use this for my bobbycar build. I think about changing my motors too delta configuration to get more maximum speed (and less torque). Is you FOC able to handle this? I see two problems (correct me if I am wrong). Commutation will change by 30deg, so no chance to fix this by changing commutation order (only x times 60deg). Maybe this can be fixed by an offset in the position estimator? With change of the motor constants maybe also some parameters of your control loop have to be adapted. Do you have an idea which parameters to scale (by root(3)?). Thanks

(for the ADCmid option I also added a deadzone option, I can make the code nice and share it with you, if you are interested)

EmanuelFeru, 2020-03-01:
hmm.. I am not sure what will happen. I did not work with Delta configuration, so I don't know exaclty which coil needs to fire, in which order.

I suggest to start with Commutation Control Type as is, and see if that one works.

Rudi84, 2020-03-02:
Ok, I checked the code and found some place where I can change my expected differences. How did you tune the control parameters for q/d. Only at the model or is there some practical observations to determine if gains are too large or too small? (have a PHD in control theory and modelling, but no practical experience with FOC).

EmanuelFeru, 2020-03-02:
I tuned is simulation and then on the read Hardware. I put the hoveboard off-the-ground and I started tuning while is running at different speeds, using [STM Studio](https://www.st.com/en/development-tools/stm-studio-stm32.html)

The main objectives were to make the motors spin smoothly and minimize noise. Sometimes I was braking it with my hand to see how it performs under some load.

Rudi84, 2020-03-04:
Ok, thanks. I will try and report, if I was succesful. But depending on time for the project, this will take some months.

Rudi84, 2020-04-22:
Hi, 
I did some more trials, also with SIN-method. Here I adapted also the sin3 table for phaseshift. Result is more clear (motor constant is different in both directions, when phaseshift is not correct). Fieldweak is also working like expected. I can nearly double speed with 25deg fieldweak for both motors start-delta (delta is reaching ~1700RPM). Measured battery current is very high. 13deg fieldweak fits better to 5A current in FOC-Mode (battery current and max speed on star motor).
In FOC fieldweak behaves totally different for delta motor. I have to increase fieldweak current times three for delta motor to get nearly same results like for the star motor. So I used up to 15A weak current, which only results in 5.5A no load battery current at full speed. Nearly same battery current is measured for star motor at 5A fieldweak. 
I also searched in the simulink model, if there is some scaling, but I found nothing. Do you have an idea for this behaviour or a point I should tune? Or is this only linked to changed motor constants by sqrt(3) or changed phase resistance/inductance? I also played with current controller parameters for Id, but nothing changed.
Best regards

gerkon56, 2020-04-29:
Hi. I also want to try to switch the wheels from the hoverboard to Delta. You need to fix the code or it will work like this.?

Rudi84, 2020-04-29:
Hi, I played a bit more. From theory higher fieldweak current is expected, but at less voltage. So field weak power stays same, but current has to be adapted. 
First driving trails showed two proplems: hard torque rippels at very low speed and noisy crying motors at half speed and full load. Both problems can be explained. First, for start block commutation is used. Here I can not compensate 30 deg phase shift, only by re-aranging the hall sensors, what I do not want to do. So I modified setting to avoid this. Starting from zero speed in FOC mode is working ok.
Second: Noise is from chopping in current limitation. Problem here that DC current is measured via shunt, so is independent from motor. But phase current calculated based on phase voltage, so depending on resistance and inductance of motor (which is changed by factor of three between delta/star). So calculated current is three times lower than in reality --> current controller is forcing too much current and power is chopped by DC current protection. I also adapted calulaton here. So it is working for me now and can be closed.

@gerkon56 : for initial trial it is working w/o changes (dont use full load, this will blow up your battery, because of this wrong current problem), but to have it correct and nice in all conditions, several changes are needed. Be aware you gain speed by factor ~1.7, but you loose torque by factor ~1.7. And I have the feeling for lower current is is getting worse. So maybe not the best way for a hoverboard where you need some margin for stabilizing control. For me I will increase the current later to compensate, delta motor can handle this, electronic with bit more cooling also. But battery is already now at the edge. For this I will add aditinal batteries in parrallel.

EmanuelFeru, 2020-05-02:
@Rudi84  very interesting and informative information on Delta motors.  It is nice to see how you tackled some of the issues.

As a general remark, if your goal is to have a fast spinning motor , from control point of view, I recommend Sinusoidal method, because it is not using phase current measurements. So, you don't face the current scaling issue.  On top, I don't know what FOC will do, since the motor has basically changed. 
Thus, for fast spinning motors, I think the best is to use sinusoidal. The only disadvantage is that there is no proper phase current protection. So, you have to rely on current chopping based on DC /Link current. 

For which application do you want to use the Delta motor?

Rudi84, 2020-05-12:
@EmanuelFeru : I also tried sinusodial. Without fieldweakening, the reach nearly same speed. And I really like the torque mode, because other modes are braking automatically, when you give zero command and that very hard. Also in Sin-Mode I reach the chopping, because the delta motor can pull more current. For FOC I also retuned some control parameters, but only for noise. Motor is working well now.

I am building an all wheel driven Bobby Car for fun. We have a small hobby Go Kart Club here and want to get the Bobby Car on a Go Kart Track.

closed 2020-10-21 by EmanuelFeru