EFeru/hoverboard-firmware-hack-FOC #28: Could you add PWM of possible remote?

BorgStefan, 2020-03-03:
Hi Emanuel,

I have taken 4 ten inch wheels and mainboard from 2 Hoverboards. Constructed my own skatebord and using Benjaf repository to make and build a software which uses PWM as remote. Hence I have a 4 wheel drive and it does around 20 km/h. 
https://github.com/benjaf/hoverboard-firmware-hack-esk8
I would be really happy if you could add PWM control to your sw. I would like to try it with my skateboard. My skateboard makes 12 km with 4 Hoverboard batteries.
Many thx for your great effort so far! 
[image]

BR Stefan

pinguinpfleger, 2020-03-04:
Hi Stefan,

PPM is already implemented. 
It is serialized PWM, means all channels goes over one wire.
You'll have to convert PWM to PPM using an arduino / esp / teensy ... 
Code you'll find here on github.

PS: interesting eboard :-)

BorgStefan, 2020-03-04:
Hi,
I am using hoverboard hack made by Niclas Fausth which Benjalf has modified to work with a PWM remote. I managed to flash to mainboards and use 2 receivers to one PWM based remote.
I will try and modify feru sw to include benjalf modifications.
A Video of my first tries with Benjalf sow.

pinguinpfleger, 2020-03-04:
I see, https://github.com/benjaf/hoverboard-firmware-hack-esk8/commit/aa910a9b9a90580f96629c2652b98f8cfc7fce3b

Thats nice and would be cool when you do this as new VARIANT_PWM

EmanuelFeru, 2020-03-04:
That would be nice to add, however is up to you to perform the testing since I don't have the HW.
I agree with Kai to add a new VARIANT_PWM to keep all the rest unaffected. If you do the changes make a Pull Request and I will review.

BorgStefan, 2020-03-05:
Hi,

Well it has been almost 11 years since I made some coding...
I am not using my skateboard for testing. I have a testbed where I try my new sw changes. Benjaf code works and the setup I made with his code is reliable but would like to try new sw.
Br Stefan

EmanuelFeru, 2020-03-24:
Done.

closed 2020-03-24 by EmanuelFeru

BorgStefan, 2020-03-24:
Oh you have done it?!
Thank you very much.
I have started on my own but then how fun it will be to try it!!!
Br Stefan