EFeru/hoverboard-firmware-hack-FOC #30: Support for YKMAX v1.0

renmalek7, 2020-03-22:
Hello. I want to flash the firmware of a pair of hovershoes to use them with a nintendo nunchuk but the main board looks different from yours.  Could you advise me please ?

Thank you

Alex

[image]

[image]

[image]

EmanuelFeru, 2020-03-22:
This board is not supported by this firmware. Only hoverboard mainboards are supported.
For hovershoes, most probably the hardware is different, pin-outs, mapping, micro-controller maybe and I am quite sure also the motor is different.

So, I think you need to do quite some work to adapt this firmware to the hovershoes. I am not saying is not possible, for sure is possible but with some effort. Unfortunately, I cannot spend time on this, so you have to do it on your own. I can support with advises as time permits.

renmalek7, 2020-03-24:
I would really appreciate your advises. I am currently living in Europe so if you have some minutes in the evening  please choose the day and time for a quick skype call. Thank you very much.

EmanuelFeru, 2020-03-25:
My time is very limited, so skype call is not possible unfortunately.

renmalek7, 2020-03-25:
I dont know where to start, I just dont want to brick the board due to a rookie mistake.
renmalek7@gmail.com

EmanuelFeru, 2020-03-26:
When implementing something new on unknown grounds it is very likely to brake something. So, it is good to have 1-2 spare boards.

About where to start, I advice the following steps:
1. Get to know your system: reverse engineer the electronics, pinouts, open-up the motor and check Y or Delta connection, count number of phases, number of poles, check Hall sensors position, measure phase resistance, etc.
2. Check the used microcontroller and its datasheet.
3. Get to know the firmware and what others did.
4. Evaluate what needs to be done to reach your goal
5. Start implementing.

I hope it helps.

renmalek7, 2020-04-06:
-I opened up the motor and the heat -shrink-tube with the wires welded together makes me think it is an Y connection.
-I dont know hot to count the number of phases.
-As seen in the photos, the number of Magnetpoles are 20 and the Statorpoles are 18.
-The Hall sensor are positioned at 0°, 60° and 120°.
-The measured resistance between phases is 0.5 ohms and 0.25 between each pahase and the center.
-This is a GD32F130 microcontroller.

Is it posible to pull the current firmware from the board so I can put it back after writing my own firmware? Any recommended procedure?
[image]
[image]

EmanuelFeru, 2020-05-02:
20 magnets motor, seems like you have a 10 pole pair motor. Try running the motor with the Commutation method, theoretically it should work.

Having a GD32F130 microcontroller, seems like similar to the sideboards. So you have to port thi FOC branch code using the same framework as I did for the GD32 sideboards, have a look [here](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD).

> Is it posible to pull the current firmware from the board so I can put it back after writing my own firmware? Any recommended procedure?

I don't think this is possible  because most probably  the chip is locked.  Unlocking it will do a full erase. But you can try.

closed 2020-07-20 by EmanuelFeru