EFeru/hoverboard-firmware-hack-FOC #36: ESP8266/Arduino control

pav18, 2020-04-07:
Hello.
Thank you for your project.
I have a problem.
I compiled online your firmvare with options UART and Torq control
I connect WEMOS D1 mini (esp8266) to UART 3 (via SoftwareSerial)
I made a sketch and send command Send(0,300)  evey 100ms.
When I send any value different from 0 the wheels start spinning. BUT the speed never changes. 100, 300, 1000 - doesn't matter. 0 - stops spinning.
What it wrong?
Please help.
Thank you in advance.

pav18, 2020-04-07:
Torq control. TORQ. When you hold wheels for example by hand, you will see speed changes. Freedrive they spins at max speed.

pav18, 2020-04-11:
I just reopened an issue to let poeople see this info.

EmanuelFeru, 2020-04-14:
Yes, this is the normal behavior of Torque mode.  It needs load... if there is no load (wheel in the air) any small torque will make it go to the max speed. So, it needs resistance to counter-act the produced Torque.
If there the issue can be closed, then please close it, because then I have in my view only the items that still need attention. Thanks!

closed 2020-04-17 by pav18