EFeru/hoverboard-firmware-hack-FOC #37: Electronic brake

pav18, 2020-04-17:
I am playing with firmware and my GoKart.
Are anybody already implemented electronic brake via Arduino?
Would you share the code please?
Just to do less experiments.
If I have torq 1000, my speed is at max and I need to fully but comfortably stop what shall I send?
If I send -1000 wheels will be blocked?  I need to increase back torq? -100, -200, -300? How to do it better?
Thank you.

EmanuelFeru, 2020-04-23:
Hi @pav18 ,

I implemented something similar in the HOVERCAR variant. Please have a look at this part of the code:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/95c10c9ba788de2bb88483276b7c03336fc31aa1/Src/main.c#L212-L233

The code above allows for smooth braking without the risk of going in the opposite direction when reaching standstill.

pav18, 2020-04-25:
Sorry, I am not so good in coding and I did not understand your code, especially it links to the functions not listed here.
I made that code below but there is a risk of going opposite direction. And also I would be happy if you tell me how to decrease the speed smooth. Also is it possible to use beeper on the main board - like horn for exmple?
Is it possible to add mode switch? to Send(Steer, Torq/Speed, Mode)  for example?
`StopC = Stop1.read();
  
  if (StopC > 0 && (Feedback.speedL_meas > 0 || Feedback.speedL_meas < -0)){
  //if (StopC > 0 && Feedback.speedL_meas != 0){
    digitalWrite(LedPin100, HIGH);
    operate = 0;
    if (gas > 60 && Feedback.speedL_meas > 0){
      gas = 0;
    } else if (gas < -60 && Feedback.speedL_meas < 0){
      gas = 0;
    } else if (StopMillis < timeNow && Feedback.speedL_meas > 0){  
      gas = gas - 100;
      StopMillis = timeNow + 100;
      digitalWrite(LedPin100, LOW);
    } else if (StopMillis < timeNow && Feedback.speedL_meas < -0){ 
      gas = gas + 100;
      StopMillis = timeNow + 100;
      digitalWrite(LedPin100, LOW);
    } 
    StopCOld = StopC;
  } else {
    if (StopC != StopCOld){
    digitalWrite(LedPin100, LOW);
    //operate = 1;
    gas = 0;
    StopCOld = StopC;
    }  
  }`

EmanuelFeru, 2020-05-02:
>Also is it possible to use beeper on the main board - like horn for exmple?

It is possible to customize the beep as you desire. As a horn, long, short beep, etc.. Just experiment with it a bit.

Regarding your code.. I don't know exactly what you want to achieve.. so I wonder is you want smooth braking why not using the HOVERCAR variant and just write  `cmd1 = Brake pedal ` and `cmd2 = Throttle pedal` from your Arduino?

closed 2020-06-23 by EmanuelFeru