EFeru/hoverboard-firmware-hack-FOC #38: High battery voltage reading on power-on

sandstreamer, 2020-04-20:
Hi,
I'm running the HOVERCAR variant and using an arduino to read FEEDBACK over serial connection. I have noticed that reported voltage (Feedback.batVoltage) when powering on the board is very high, like 400V, and then it gradually gets down to normal levels after 30 seconds or so. 

Do you know if this is normal behaviour or perhaps an issue with hardware or software?

Thanks for this excellent project!

yashatan, 2020-04-22:
I have same problem too. And i think this is normal behavior. It takes  about 20 secs to gets down to right value.

EmanuelFeru, 2020-04-23:
Hi,

Yes, that is a normal behavior. Initially, we don't know what the battery voltage is. So, we start with a guess of 40V (which is 4000 in the firmware) and converge to the real voltage.

Another way would be to save the battery voltage before shut down and then read it back at the next power on... but then the flash memory will wear out in time due to many writing cycles. So, I don't recommend.

sandstreamer, 2020-04-23:
Thanks for the explanation! Closing the issue.

closed 2020-04-23 by sandstreamer