EFeru/hoverboard-firmware-hack-FOC #39: Movement less than one turn with reverse

albertgranya, 2020-04-23:
Hello Emanuel,
First, congratulations for nice code,

After some try catch with the code and one hoverboard, I start realize your project.
This is about limitations on FOC and low speed, and I understand than at low velocity is moving by torque , and with enough speed FOC starts.
The motor has differential encoder, and I need an extra sensor for position zero degrees, obviously with offset to proper setup, but differential encoder hasn't enough precision for my application.
No big load attached to the motor, minus than the weight of the tire, and the tire is not present, for this application.

I need the precision on rotor angle, but the movement I need for my application is a little bit strange:
The movement starts at 0 degrees.
I need to start with velocity enough to start FOC properly, and synchronize positions.
The move reverse at 180 degrees after and comebacks to 0 degrees.
Some accelerations and decelerations for smooth move.

The precision is important only near 180 degrees, in fact the target is in range between 160 and 180 degrees.
I understand than in the reverse movement, velocity comes to zero,

The question is:
Is possible to use FOC angle rotation, in this move and get enough repetitive on the move,mechanically?,
that we don't know exactly where position it really stops,

If the answer is yes, which kind of precision you belibel I get with FOC angle rotation?, if compare with differential encoder, understanding than one AD converter is not perfect, and the electronics made in china is far perfection.

If the answer is not,

Focus on the reverse move,
Target 170 degrees.
With FOC angle rotation before low velocity,
Disconnect FOC
start reverse by torque,
understand than now, we don't know where is the rotor by FOC,
differential encoder is not used yet, only after that to get zero degrees position.

Do you thing than this can has repetitive mechanically?,
everything and knowing that we don't know exactly where position it really stops,
and reverse.

Thanks for your time and great job with this hack, is not common, congratulations .

pav18, 2020-04-26:
Hello.
I would appreciate if you share info how to increase the speed. At present time I have approx 18 Km/h  with 350Watt wheels and 36V battery

EmanuelFeru, 2020-05-26:
Hi @albertgranya,

Sorry for my late reply, Your application seems quite demanding for what a hoverboard can do. From your description, I see you need accurate position control, unfortunately the hoverboards are not designed for position control but rather  for  continuous motion, without much attention on position.

Therefore, it is quite challenging to do position control with  just using the existing Hall sensors. If you want to do proper position control then you need to connect an encoder to have precise position of the motor.

Someone did position control using a magnetic encoder. This is the Youtube movie:
https://www.youtube.com/watch?v=aRjTd3s7-0Y

closed 2020-07-20 by EmanuelFeru