EFeru/hoverboard-firmware-hack-FOC #42: Turn On/Off

pav18, 2020-04-26:
I use two hoverboards in my 4WD project. I control boards via ESP32. Use 2 UARTs. When I push power button at main board (where ESP takes power), the second board also switches on (surprise). BUT when I push button again only main board switches off but second still working.
I think that everything could be controlled software.
Is it possible to add feature to turn off second board when main board receive command to switch off (hard button)?
Is it possible to implement UART switch OFF?
Thank you.

yashatan, 2020-05-01:
> Is it possible to implement UART switch OFF?

I think yes it is. You can call [poweroff()](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Src/util.c#L547) function to switch off your board.

EmanuelFeru, 2020-05-02:
Thanks a lot  @yashatan  for support.

pav18, 2020-05-05:
> Or just simple connect 2 button port of both to ONE button

is it safe?

pav18, 2020-05-05:
> Thanks a lot @yashatan for support.
@EmanuelFeru is it possible to add turnoff, beep to default UART firmware. Friendly speaking I am not a programmer at all and all this look a little bit complicated for me. I use online compilator for your firmware and like it very much. But it would be nice if you put everything inside the firmware. Best,

yashatan, 2020-05-06:
@pav18  do you have telegram or whatsapp, i will help you code it

pav18, 2020-05-06:
> @pav18 do you have telegram or whatsapp, i will help you code it

+7 nine 1 seven ... 5 ... seven 9 8....five six... 5   I would appreciate your help.

yashatan, 2020-05-06:
Why don't you just type your number? why "nine" "seven"...?

pav18, 2020-05-11:
@yashatan  not to let searching machines index it ;)  Both and whatsapp and telegram

yashatan, 2020-05-12:
ok i have add you

closed 2020-06-03 by EmanuelFeru

waythe, 2022-01-22:
hi @pav18 , 
have you finished your project and would you mind sharing some experience since I am just having the same trouble with you.
I've written my problems in https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/256