EFeru/hoverboard-firmware-hack-FOC #44: Hoverboard firmware

moezmahroug, 2020-04-29:
Hi,
I'm a new member of this wonderful forum. I would like you to help me find a solution to my problem, please!

So, my problem is that I purchased Hoverboard for robot projects. Unfortunately, I discovered that it is not original. My project is to control their motors with Arduino. So, I tried to work on it, but I made a mistake and I lost the firmware. There are three CPUs and I only work with one of them. I will mention it in the pictures.
If you can help me get him back to work as they were or given me a solution to control it with Arduino, I would be very grateful.

type of board => YK95 V1.1

[image]
[image]
[image]

EmanuelFeru, 2020-05-02:
Once the firmware is erased, there is no way to recover it (as far as I know).

The board you have is not a standard board, so I don't know what will happen if you use my FOC firmware. However, I have seen people compiling the firmware on such boards, but don't take it as 100% true. I think you should check the microcontroller pinouts before flashing to avoid breaking stuff.

moezmahroug, 2020-05-02:
Hi,
Thanks a lot for the quick answer.
This is my mistake, I tried to open its protection, but the memory was erased. I think I will try to drive the MOSFET drivers at first if it's ok a will move on to get signals from hall sensors.
I will try to find a program running on Arduino to control the MOSFET drivers. If you have any links for a program I can use it.

Something like this in the picture : 

[image]

closed 2020-05-02 by moezmahroug