EFeru/hoverboard-firmware-hack-FOC #45: USART version won't upload to the main board.

odox, 2020-05-01:
Hi Emanuel!
I use your non-FOC version for my robot lawnmower project https://www.youtube.com/watch?v=JAYIfTFY9Fk&t=7s . And it work as a charm. However, I saw you added speed feedback mode in the FOC-version and I wanted to try that for my mower, but the USART- version won't upload to the hover-main board. I't gives a continuous beeping every second. The LED on my ST-LINK V2 toggles between blue and red after the upload attempt. I tried the ADC-version and it uploaded without issues.

EmanuelFeru, 2020-05-02:
Hi,
Nice project! Looks really good. Do you use any navigation sensor?

Regarding the USART variant, the beep is from Serial Timeout. The board expect Serial values as soon as it is turned on. If values are not received (or not received correctly) it enters Timeout mode and emits the beep that you hear.

odox, 2020-05-02:
Thanks! I'll try out the USART-thing tomorrow. I'll let you know...

I was testing the invisible fens in that video clip. There is a wire buried in the grass that sends a  binary code. In front of my mower is a coil connected to an analog input on my Arduino Mega board via an amplifier. Is sense the code and if it change the polarity it knows it is outside the wire and tells the motors to reverse and turn. I will add more sensors later. 
I got the perimeter wire setup from here: https://repalmakershop.com/pages/perimeter-wire-module witch is from the Ardumower projekt. I have not find any one using hoverboard wheels witch in my opinion is the best solution for a mower. Silent and extremely strong:-)

EmanuelFeru, 2020-05-03:
Interesting idea! Thanks for sharing. 
I agree that hoveboard motors are very strong indeed and can be used for many applications. That's why I like them too.

closed 2020-10-21 by EmanuelFeru