EFeru/hoverboard-firmware-hack-FOC #46: Sinusoidal vs FOC, voltage vs speed vs torque

juodumas, 2020-05-02:
Hey @EmanuelFeru,

Thank you very much for all your work.

I have been driving my tricycle (two 10" wheels on the back axle) for a while now using bipropellant firmware with your sinusoidal motor control. It has been great compared to the commutation method, more efficient and better max speed (40kmh with 16x LiFePo4), and no issues over 100km+. The only thing where I wished for improvements was quite noisy current limiting.

Yesterday I had time to try this FOC firmware hoping for improved battery efficiency. I chose voltage mode and drove for 10km. Current limiting is working great and not too noisy, but at one point when trying to drive up a 10cm barrier, motor diagnostics decided that motor is stalled and turned everything off. Will have to experiment more with this. Max speed was very similar to sinusoidal. When wheels were not touching the ground, sinusoidal at 500rpm used 1.1A, FOC at 500rpm used 0.9A. But I could not confirm better battery life because battery died much sooner than with bipropellant+sinusoidal. Will test more. I am using ADC control method and was hoping maybe you could help with some questions?

1. Does torque mode use regenerative breaking? Couldn't really test it yet because ADC channel picks up more noise in torque mode, which is strange (I have 100K resistor between ground and ADC signal).
2. Voltage/speed modes: motors are locked in position when stopped, which is different from previous firmwares. I think there should be an option to disable motors when not moving and no intention to move. I have added this condition to main.c loop - it keeps motors disabled if true: `enable == 0 && speed < 20 && speedAvgAbs < 300`. But I think it's specific for my use case and this could be somewhere in motor control? I like to be able to freely move my tricycle using my feet for a meter or two.
3. Now there are three current limiting variables: I_MOT_MAX / I_DC_MAX / FIELD_WEAK_MAX. Does FIELD_WEAK_MAX have any influence on max current use? Or will current use always be below I_DC_MAX, independent of FIELD_WEAK_MAX setting?
4. When experimenting on a test stand, max rpm (`rtY_Right.n_mot`) I could reach was about 850. As far as I see, current limit wasn't reached and I have printed these values: `adc=4095 cmd=1500 speed=1467 rpml=0699 rpmr=-854`. Later I have confirmed with GPS that max speed is about `40kmh (850rpm)` when driving. Of course it's more than enough for my use case, but I am curious, what could be limiting max rpm? I have tried upping these values: `FIELD_WEAK_MAX 7`, `I_DC_MAX 17`. What is a safe limit for `FIELD_WEAK_MAX`?
5. If I want to compare sinusoidal vs FOC battery efficiency, do you think it's safe to use this firmware with sinusoidal mode or better to revert back to bipropellant+sinusoidal?

Thanks again!

EmanuelFeru, 2020-05-03:
Hi, thanks for trying out this repository! Here are my answers:

> 1. Does torque mode use regenerative breaking? Couldn't really test it yet because ADC channel picks up more noise in torque mode, which is strange (I have 100K resistor between ground and ADC signal).

All modes use regenerative braking. Once the current changes sign, the power is going into the battery. For karts, tricycles, or anything that you sit on, I recommend Torque mode, it gives a much better driving experience, plus you have freewheel at 0 torque. I use torque mode also for my Hovercar and I am very happy with it, see video here:
https://www.youtube.com/watch?v=IgHCcj0NgWQ&t=

> 2. Voltage/speed modes: motors are locked in position when stopped, which is different from previous firmwares. I think there should be an option to disable motors when not moving and no intention [...]

This is a normal behavior of Speed mode and desired sometimes. For example, someone in the [Telegram group](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw) uses speed mode to lock his Bobby car, to prevent someone from driving away or pushing the vehicle. So, I would keep any very specific implementation to the user and not put it inside motor control. I advice if you want to move the vehicle freely, either disable the motors like you did, or change the mode to Torque mode.

> 3. Now there are three current limiting variables: I_MOT_MAX / I_DC_MAX / FIELD_WEAK_MAX. Does FIELD_WEAK_MAX have any influence on max current use? Or will current use always be below I_DC_MAX, independent of FIELD_WEAK_MAX setting?

The current limitation includes the Field Weakening (if active). So, even if Field Weakening is active it will not draw more current from the battery than the I_MOT_MAX. The I_DC_MAX is used as the 2nd stage current protection and for the Control types that do not use phase current measurements (like Commutation or Sinusoidal).

> 4. When experimenting on a test stand, max rpm (rtY_Right.n_mot) I could reach was about 850. As far as I see, current limit wasn't reached and I have printed these values: adc=4095 cmd=1500 [...]

I think the max speed that can be reached with this motor is given by its construction, battery voltage and battery current. To be honest, I do not know what is the ultimate limit for FIELD_WEAK_MAX. I think we have to find that by experimenting. During my trials I did go up to 10A with my motors and they are still working fine.  But again I don't know what is the max limit. I think at some point the max speed that can be reached will saturate, and anything above that FIELD_WEAK_MAX will just have a penalty in torque. These values for sure can be calculated mathematically, but I didn't have time to spend the effort for it.

> 5. If I want to compare sinusoidal vs FOC battery efficiency, do you think it's safe to use this firmware with sinusoidal mode or better to revert back to bipropellant+sinusoidal?

From motor control point of view, I recommend this version. I improved significantly the Sinusoidal method here compared to bipropellant: it is computationally much faster (because of fixed-point calculations) and has some functionality updates. So, I recommend this one.. unless you miss some functionality from bipropellant, then you can still use bipropellant.

I hope these answers help.

juodumas, 2020-05-03:
Thank you, very helpful.

Torque mode sounds interesting, but I have no separate brake pedal/handle. So I depend on regenerative braking to stop, even when torque is 0. Is there a way to prevent torque mode from freewheeling when torque is 0?

In other repositories motor is _not_ locked into position when cmd1/cmd2 is 0 and not moving. Do you see any simple way to modify this repo to make a real drop-in replacement? Maybe only for voltage mode? For example, how do you make an electric scooter which has only a throttle handle and no brakes besides regenerative?

Yesterday I drove another 12km with FOC + voltage mode. The board has turned off again a few times as if battery was dead, without any beeps, the wheels just started freewheeling and I needed to power on again. I am not ruling out battery issues just yet, but it happened at ~4km and then I drove further 8km...

Planning to compare sinusoidal vs FOC, will share i get any interesting results.

Thanks!

EmanuelFeru, 2020-05-03:
> Is there a way to prevent torque mode from freewheeling when torque is 0?

There a 2 ways to do it:
1.  To apply some negative torque (calibratable to your needs) when `speedAvgAbs > 0 && Throttle == 0` -> advisable to use Thresholds rather than crude `0`.
2. Use a button to reverse the torque sign. You can use the other ADC channel as a button, and when pressed it reverses the Torque sign. Like this you are able to brake proportionally and also drive reverse, while option 1 can't do.

Regarding the battery issue shut down, did you calibrate you ADC battery levels in config.h?

juodumas, 2020-05-03:
Nice, so if I set pwml/pwmr to `-1`, then regenerative breaking will kick in?

> Regarding the battery issue shut down, did you calibrate you ADC battery levels in config.h?

Ok, it's a bug I introduced while modifying for single ADC control :( Sorry I bothered you about this.

EmanuelFeru, 2020-05-03:
I advice to change the sign of `cmd2` (Throttle). If you set directly pwml/pwmr then you bypass all filters and that could lead to a too sudden torque change resulting into a shock.

I don't think the battery life is affected by using regenerative. The BMS should take care of managing this free energy and store it properly into the battery.

juodumas, 2020-05-05:
I think I am misunderstanding torque mode.
For example when motors are spinning at 300 rpm, if I change pwml/pwmr (after filtering) to 0, motor speed does not change. Why?

slip11, 2020-05-13:
Hi, I also made a scooter on your firmware. Everything is fine, thank you very much.
If you go with a speed of 23-42 km / h,(10') the freewheel does not work, if you throw the throttle stick, the wheels will roughly brake to 25. I wanted to change RATE and SPEED_COEFFICIENT but there was no result. It’s scary to ride when the controller tries to throw it overboard =)
Parameters: FOC, TORQUE modes, FIELD_WEAK_MAX 9, I_MOT_MAX 14, N_MOT_MAX 870

EmanuelFeru, 2020-05-14:
Hi @slip11 , can you please explain the situation in more detail: 
- what do you mean by "throw the throttle stick"? It is not clear to me
- what VARIANT_* are you using? and what is your input method ADC, or UART or.. other?
- do you have Field Weakening enabled?
- when the braking happens, do hear any beeping?
- And does it recover to normal driving after some time? Is there a need to stop to standstill or restart the board to recover?

Just for your info, I drove with my Hovercar last Sunday in Torque mode with Field weakening, and I didn't experience issues.

EmanuelFeru, 2020-05-14:
@juodumas , that is a normal behavior of Torque mode, if motor Torque is higher than the Motor friction torque and Wheels are off the ground (with no load) the motor will just rev up to max speed.

Once there is load (for example wheels on the ground) it will not rev up due to the road.

slip11, 2020-05-14:
Thanks for the answer. variant_Hovercar, Field_Waking enable, the board continues to work without problems.
what I managed to find, when the wheel speed is more than 480 rpm, the freewheels do not work.

EmanuelFeru, 2020-05-14:
you mean that above 480 rpm, went you put the torque input to 0, it won't freewheel?  Does it brake quickly when speed is above 480 rpm? any beeping?

slip11, 2020-05-14:
480-850 rpm only if SPEEd in serial port >1000, everything works correctly, as I understand it #define  DEFAULT_RATE 480 needs to be changed

juodumas, 2020-07-19:
My side of the questions have all been answered, so I am closing this issue. Huge thanks @EmanuelFeru.

I haven't tried FOC yet. Tried experimenting with the code in main.c and burned my board while braking, most probably because of bugs in my code or because of full battery.

Another spare board that I had is with AT32F403 CPU, so I am currently using it with sinusoidal + voltage. It works perfectly with impressive top speed, but unfortunately current limiting does not work, so I am not sure if it's safe to introduce support for this board to the main repo? When I get a chance, I will try FOC with the changes in #81.

closed 2020-07-19 by juodumas

remitoutsimplemnt, 2020-09-10:
hi @slip11 , do you have found solution for your problem?

I have same and that is really problematic !

slip11, 2020-09-10:
No. Field_weaking <5 use

remitoutsimplemnt, 2020-09-10:
and how fast can you go

slip11, 2020-09-19:
32