EFeru/hoverboard-firmware-hack-FOC #49: Sometimes when starting from a standstill motors need to turn for a few rounds until they start responding normally

juodumas, 2020-05-03:
Today I tried sinusoidal type and noticed one difference (issue?) compared to sinusoidal in bipropellant. Sometimes when starting from a standstill, I have almost no acceleration and the wheels sounds like they are at current limit. Max speed is <10kmh, the wheels are screaming and the only way to fix is to lift the wheels above ground and let them turn at higher rpm. Happened three times with sinusoidal and something similar happened once with FOC. But with FOC the issue fixed by itself after driving for a few meters.

Not sure how to reproduce yet. I think this happens when motors are disabled (`enable=0`) and wheels move for a few rotations. Will update then figure out a more reliable way to reproduce.

EmanuelFeru, 2020-05-03:
The motors should have high torque from 0 speed. Seems like an issue with motor phase mapping. Have you tried swapping the motor phases?

juodumas, 2020-05-20:
Closing because I did not reproduce it yet.

closed 2020-05-20 by juodumas