EFeru/hoverboard-firmware-hack-FOC #51: How to calibrate?

KaanDemirkoparan, 2020-05-14:
Hi. I really appreciate you, that is a really nice work. Thank you for that.
Currently I am making my own e scooter. I am a prep student so I don't know much about firmwares and codes. There are some points that I couldn't understand. Can you help me with those?

1) İnside the config.h there are some variables about battery. In the comment section you explained calibration of the battery. But I couldn't understand the explanation. What does "see How to calibrate" means? Is there any explanation about how to calibrate? If there how can I see it? And also what does nr 5 and value 6 means? How can I see them?

2) On my scooter I will use a key switch as on off button so it will not work like on off button (press once and release turn on, press once and release turn off).  When the key switch is at the on position my board should work, when it is at off position board should be turned off.
 
Those are my questions for now. İf I have new questions can I ask? Thanks.

[image]
[image]

yashatan, 2020-05-14:
hmmmm. With your key i think you have to make a lot of adjustments. Maybe you should design an digital IC circuit to get rising and falling edge to make an high signal.

> İnside the config.h there are some variables about battery. In the comment section you explained calibration of the battery. But I couldn't understand the explanation. What does "see How to [...]

Step 1. Charge your battery full. Measure it by VOM and multiply by 10. It is BAT_CALIB_REAL_VOLTAGE value.
Step 2. Turn on your board, print out batVoltage value via UART, then wait for it converges. It's BAT_CALIB_ADC value.

KaanDemirkoparan, 2020-05-15:
Thank you for your answers. 
I have to think about switch. Worst scenario would be using key switch and button at the same (connecting them series). 
With your answer I understood the calibration procedure. But as I mentioned earlier I don't know much about codes and firmwares. So I don't know how to print out values via UART. Can you help me about it?
Thank you again.

yashatan, 2020-05-15:
OK. Do you have Telegram or Whatsapp?

KaanDemirkoparan, 2020-05-17:
Yes I have whatsapp.

closed 2020-05-26 by EmanuelFeru

pav18, 2021-02-25:
Sorry Gents but I have to open the issue again. I understand that everything looks easy when you did it already but instructions are not clear enough.
What I did. Default battery settings.
DEBUG_SERIAL_USART3 uncommented
Battery is fully charged
Measured voltage by VOM - 4130
Get debug info - adc_buffer.batt1 - 1630    (batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC))  - 4334
Changed BAT_CALIB_ADC to 1630
Changed BAT_CALIB_REAL_VOLTAGE to 4130
Uploaded firmware
Get debug info - adc_buffer.batt1 - 1630    (batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC))  - 3962
WTF????

I even do not clarly understand calibration process.
(batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC))   =  X * 4130 / 1630  = 3962.     batVoltage = 3962*1630/4130 = 1563. WHAT is batVoltage?????

What is wrong here?
Please help.
Thank you ;)

vamfun, 2021-02-26:
I’m not an expert in this but I believe you are correct.... the formula should read
BatVoltage = adc_buffer.batt1*BAT_CALIB_REAL_VOLTAGE/BAT_CALIB_ADC

Somehow the printf scope variables got modified.  This was the equation in original NF hoverboard hack firmware.

Maybe Emanuel can weigh in.

http://vamfun.wordpress.com
Robot musing blog
iPhone

yashatan, 2021-02-26:
@pav18 

You have done calibration correctly. Now i will explain you why.
But wait. What is 4334 and 3962 you are talking about? Why you calculate it? I even don't understand what it is.

> Get debug info - adc_buffer.batt1 - 1630 (batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC)) - 4334

> Get debug info - adc_buffer.batt1 - 1630 (batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC)) - 3962

vamfun, 2021-02-26:
I believe he was substituting actual voltage for the variable batVoltage.    That was the confusing factor.   See my edited comments on my last post.

pav18, 2021-02-26:
> But wait. What is 4334 and 3962 you are talking about? Why you calculate it? 

Well. I get debug info first time -4334. Then changed Changed BAT_CALIB_ADC to 1630 according instruction and changed measured acc real voltage.
Uploaded firmware
Get debug data again and got Get debug info - adc_buffer.batt1 - 1630 (batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC)) - 3962

What is not my real battery voltage 4130.

I thought that after I entered my real meaured acc voltage and  calibADC which I got from debug data, debug data shall change and show something like adc_buffer.batt1 - 1630 (batVoltage * BAT_CALIB_REAL_VOLTAGE / BAT_CALIB_ADC)) - 4130

But not.

vamfun, 2021-02-27:
I think you just have to let the battery filter reach its steady state.    Looking at the code...the filter is initialized at 40*CALIB_CAL/BAT_CALIB_REAL_VOLTAGE which in your initial measurements would be 40*1630/4130= 1504 ....The filter will eventually reach 1630 but any debug readings before steady state is reached will be somewhere between 1504 and 1630.