EFeru/hoverboard-firmware-hack-FOC #54: Need to add a friction coefficient

cohnhead66, 2020-05-25:
I am using a hoverboard motor for a sim racing direct drive steering wheel.  Using ADC torque control works great except when I  let go of the wheel, it oscillates badly because there is no friction to slow down any command given.  Thinking it would be pretty easy to add a torque coefficient, just not sure how and where.

Basically cmd = cmd - (velocity /friction)

any advice on the best way to do this?

EmanuelFeru, 2020-05-26:
Let me see if I understand correctly, when you hold steering with torque is ok, but as soon as you release because of accumulated torque and no friction it starts spinning. And then because probably you have some limitations it jumps into the other direction.

Basically what you have is an under-damped system. So, a damping factor needs to be included. Thinking quickly, is something you suggested already:

`cmd = cmd - gamma * velocity`, where `gamma` is the damping factor that needs to be tuned to avoid oscillatory behavior. 

I am not 100% if this will work, because at low speeds it is hard to have a good estimate of the speed because there is no encoder on the motor and we rely only on the Hall sensors. But it worth giving a  try.

cohnhead66, 2020-05-26:
Thanks for the response.  

If I understand the code correctly, I should add this in the main loop right after the readCommand() and calcAvgSpeed(), that way the commands will still get processed through the filters and mixer.

I will give this a try tonight and see what happens.

EmanuelFeru, 2020-05-26:
> If I understand the code correctly, I should add this in the main loop right after the readCommand() and calcAvgSpeed(), that way the commands will still get processed through the filters and mixer.

Yes, that is correct.

cohnhead66, 2020-06-01:
I was able to implement.  Helped a little, but not as effective as I had hoped.  Still very unstable with no load.  I solved this by adding a small amount of mechanical friction to my system to knock down the oscillation.  If possible it would be great to add a damping function, but if not, the mechanical solution works for me.

This issue can be closed.

closed 2020-06-01 by cohnhead66