EFeru/hoverboard-firmware-hack-FOC #55: Motor Calibration Process

cohnhead66, 2020-06-01:
Emanuel, Thanks for all your help.  My project of building a direct drive force feedback steering wheel for racing simulator is working well.  However, I would like to reduce the "Cogginess" of hoverboard motor that I can feel in high torque, low speed situations.  I am using torque mode for this application.

Is there a method to calibrate/commutate the offsets between the hall sensor position to the motor poles?  I can't find anything in the documentation.

Would sinusoidal be better then FOC for this application?  Have plenty of torque, just need smoother motion.  

Thanks

EmanuelFeru, 2020-06-01:
Thanks for the feedback on the friction coefficient.

- Regarding the cogginess, it is because at low speed I use a combination between Commutation method and FOC. That is because I cannot rely on angle estimation to use full FOC. So, you still hear the Tick-Tick noise from commutation. Is it disturbing or affecting you? Because the factory hoverboard firmware does similar.

- For the Hall sensors, I didn't consider Hall sensor offsets. I agree that probably they are not aligned perfectly, but I didn't encounter issues because of that.

- For your application, I don't think Sinusoidal is suitable, because it will be more difficult to get smooth motion and the behavior is maybe not what you want. You can give it a try, but I fear that when you will send a command it will maintain a speed making the steering wheel spin continuously without being you able to stop it. While Torque mode gives a Force, which you can counteract with your hands.

cohnhead66, 2020-06-03:
I do have an 600 ppr optical encoder attached to the wheel with a 2:1 gear ratio, caculating to 4800 counts  per wheel revolution.  Would it be possible to use encoder instead of the hall sensors to improve angle estimation and all full FOC.

How difficult would that be to implement?

EmanuelFeru, 2020-06-03:
It is possible but it would require some effort because, first I need to create that bypass in the Controller, and then some calculation is needed to convert the sensor values into shaft value, with proper ratio 2:1 and offset.
It is possible, but I have to spent some effort on it and generate new code to add this feature. 

I will put a note, and depending on my time I can maybe create that bypass for you to have the angle signal as input. It will be up to you to convert the encoder values to motor electrical angle. But it shouldn't be that difficult.

closed 2020-10-21 by EmanuelFeru