EFeru/hoverboard-firmware-hack-FOC #56: Steering not working on PWM

emilhse, 2020-06-02:
Hi everyone, I'm new in all of this. i tried PWM version I compiled successfully but no mater what i do steering is not working. can you please advise me how can fix it?

benjaf, 2020-06-02:
The code for PWM steering seems to be based on the wrong version of my fork. The working PWM steering is found in this version of [control.c](https://github.com/benjaf/hoverboard-firmware-hack-esk8/blob/RC-Steering/Src/control.c).
You should be able to just overwrite the whole CONTROL_PWM section (line 118 onwards).
I would submit a pull request, but I have no way to verify if the code i actually working right now.

EmanuelFeru, 2020-06-02:
@benjaf  it would be great if you can provide a pull request.  I might have ported the wrong code for the steering channel.

benjaf, 2020-06-02:
@EmanuelFeru I got the RC receiver working with my gokart so the changes could be tested. I have submitted a pull request with the fixed version!

EmanuelFeru, 2020-06-02:
Awesome! I will have a quick look and merge. Thanks!

EmanuelFeru, 2020-06-02:
@emilhse is fixed. You can give it a try. Thanks a lot @benjaf !

emilhse, 2020-06-02:
Thanks a lot gentlemen !!!!!!!!!!!

closed 2020-06-03 by EmanuelFeru