EFeru/hoverboard-firmware-hack-FOC #58: Cannot get speed mode to work

jpiat, 2020-06-02:
Hi,

i'm trying to control the hoverboard (10" hoverboard)  motors through a serial link in python. It works fine in voltage mode but when trying to use the speed mode (wheels off the ground for safety), whatever speed value i send (sending speed of one), the motor revs up to a high speed (lot higher than what i expect). Is the USART control compatible with the speed mode ? Any idea of what is causing my issue ?

thanks

EmanuelFeru, 2020-06-03:
Hi,
The USART should work with all the modes. Can you check that the input on the mainboard is received correctly? Maybe you can write it on the Serial Debug on the other UART cable.

jpiat, 2020-06-03:
I'll try debugging on the other UART but the data seems good since i'm able to control in voltage mode. Is it possible that my Hall encoder wiring is different ? One things that looks weird to me is that when setting speed (in voltage mode) to a positive value, one motor report a positive speed and the other a negative speed. 

Here is the output of the feedback (with the sync symbol 0xABCD):

(0, 150, 118, -116, 3831, 355, 0, 23349)

i'am driving the motors with the steer at 0, speed at 150. If i increase the speed, th motor speed increases. With speed mode (what is the unit of speed in that case, RPM ?), the motor jump straight to high speed and causes the controller to reset.

EmanuelFeru, 2020-06-03:
- It is ok to have one motor positive and one negative because this is how they spin (one is mirrored compared to the other).
- Does it spin smooth in Voltage mode from 0 to max Input and Reverse also? If yes, the Hall sensor wiring should be ok I think.
- For speed mode the input is in RPM by default. The 1000 rpm is coming from here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/558759b2c5873de84fbc5f6a5d0c324db34c8e22/Inc/config.h#L135
You can increase speed resolution by lowering the number above, then the input will not be rpm any more but a scaled value between 0 and what you set above. For example if you set `N_MOT_MAX = 600 ` for an input of 1000 sent on ADC or UART, the firmware will request 600 rpm from the motor.  I wrote this just as info, but I guess 1 rpm resolution is more than enough, so you can leave that number unchanged, unless you want to limit the maximum motor speed for some reason.
- Still, I don't understand why you have issues with Speed mode, because I just tested in in my [Hovercar](https://www.youtube.com/watch?v=IgHCcj0NgWQ&t=) which uses ADC input and it works without issues. Do you maybe have any changes made, or you use unchanged FOC branch?
- it shouldn't make a difference but what MCU do you have GD32 or STM32?

jpiat, 2020-06-04:
Hi,

thanks for your feeback. Using the technique of manually setting the speed to a static value in code (50) and setting one motor to 0 speed (speedL or speedR = 0), i managed to fix my hall sensor wiring. For some reason, the wiring was not the same as the one expected in your code. Now it works great ! Thanks

EmanuelFeru, 2020-06-04:
Great!

closed 2020-06-05 by EmanuelFeru