EFeru/hoverboard-firmware-hack-FOC #59: Controlling 350W Motors

blubbi321, 2020-06-03:
Is it possible to drive the motors of "offroad" hoverboards with this driver as well? They seem to have the same connectors but I am wondering since all the hoverboard hacks seem to revolve around the 250W versions ..

Chaos99, 2020-06-03:
My original hoverboard was labelled as 2 x 400 watts. Both motors and electronics work well with the firmware.
But those maximum ratings for the motors are probably bogus anyway. The battery is not able to supply this kind of power.

EmanuelFeru, 2020-06-03:
I agree :)

blubbi321, 2020-06-03:
Thanks for the rapid replies! ;)

Ill rephrase: Do I need to change any parameters in case I wire the board to different motors from different hoverboards? (Excuses, Im pretty new to motor control)

EmanuelFeru, 2020-06-03:
Then it could be that the motor design was changed and that needs to be checked.
I am not sure if you can open up the motor and check if you have the same number of poles  and magnets and the hall sensors are positioned in the same location.

Or you could do trial and error, connect the board to the motor and apply very small input and check if the motor spins smoothly. You might need to swap motor phases or even Hall wires around, sometimes they are mapped differently from hoverboard to hoverboard.

Chaos99, 2020-06-03:
The setting you should look into is I_DC_MAX in config.h
By default it's set to 17.
2*300Watt / 36V is roughly 17A
If your motor is rated with less than 300Watts, you should decrease that value. If your battery (not your motor) can supply higher current, you can increase that value.
A bigger motor (like mine with 400Watts) just gives you a bigger safety buffer. Higher weight doesn't necessarily mean higher rating. The additional mass needs to be from more copper windings. Weight of magnets or steel frame doesn't count.

benjaf, 2020-06-04:
I have seen the '250w' motors driven at 600w without issues. There seems to be a few different versions of the motors, some with beefier windings then others. They generally seen to handle quite a bit more than their rated power though!

blubbi321, 2020-06-04:
Alright, thanks for the input. 

@Chaos99 the problem Im looking at is that these "bigger" motors dont seem to output the whole torque that Id be expecting .. so I guess Ill check the output of the battery first. Do you have any rough values about how many amperes the hoverboard packs should be able to provide?

benjaf, 2020-06-04:
@blubbi321 The packs I have will not output more than ~10A before the voltage drops significantly. This is heavily dependent on the battery pack though, so your mileage may vary. Some hoverboards seem to come with LiPo packs rather than 18650 based ones - those might be more capable, but I have yet to come across one.

blubbi321, 2020-06-06:
I see. The reason I am asking is that I tried to reproduce the values of this guy here [1]. He seems to get around 12Nm worth of torque out of the "standard" (250W) motor.
Using a similar setup I was only able to get around 5-7Nm out of both motor types.

Id need higher torque (the 12Nm [1] got would be sufficient) for my application. Any ideas on what to change in the driver? Or is it a matter of using a completely different driver (hardware) at all? 

[1] https://www.youtube.com/watch?v=OgrCXGPbfkk

EmanuelFeru, 2020-06-12:
@blubbi321 you can increase the motor allowed currents to get more torque
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/558759b2c5873de84fbc5f6a5d0c324db34c8e22/Inc/config.h#L133-L134

blubbi321, 2020-06-12:
I saw that but didn't dare to touch it. Any ideas on what is still ok there in terms of amps?

EmanuelFeru, 2020-06-13:
I think 25 A/motor should still be OK. I don't know what torque do you get for that current though.

blubbi321, 2020-06-15:
Alright, thanks again for all the input. I got (briefly) the same values as the guy in [1] above. So roughly 600W gives me ~12Nm

closed 2020-06-15 by blubbi321