EFeru/hoverboard-firmware-hack-FOC #60: simple firmware to flash with stm32cubeprogrammer

mihaigabriel620, 2020-06-06:
hello i want a firmware with foc control to use on my hoverboard with no mods
just as a hoverboard
and to have all 3 modes selectable by pussing different times the power button
and ho hold the power putton to power off
pressing will just change the modes
PLS DO THAT

Chaos99, 2020-06-06:
Commutation and sinusodial (which defaults to commutation for slow speeds) will not work well for a balancing applications. I will offer a pull request for balancing in FOC mode in a few weeks, but it will not be a ready-made solution for hoverboards. More work need to be done for this.

mihaigabriel620, 2020-06-06:
Thanks i hope it will be possible

EmanuelFeru, 2020-06-06:
Balancing control is not yet available. Bipropellant did some work in that direction and some other people as well. So, fell free to also contribute in this direction.
The features that you want related to the buttons are too application specific, so you have to implement them yourself for your own application.

@Chaos99 I didn't know you are focusing on balancing controller. Nice to hear, I think is an interesting control problem to solve.

mihaigabriel620, 2020-06-06:
Im a noob to coding i just have the stm32 snd the software

closed 2020-10-21 by EmanuelFeru