EFeru/hoverboard-firmware-hack-FOC #62: VARIANT_HOVERBOARD

vfear, 2020-06-06:
And one more question for another project. How does VARIANT_HOVERBOARD work? I want to make an onewheel project(https://onewheel.com/), will this option work for me? There's a caveat the Board should only start to keep balance when both feet are on the Board.

EmanuelFeru, 2020-06-06:
That variant is in progress. The balancing controller is not yet implemented. It is planned to be implementer but I cannot give you a date. All the communication with sideboards is done, only the balancing stuff needs to be worked out.

vfear, 2020-06-06:
Thank you, I'll be waiting =)

Chaos99, 2020-06-07:
You can have a look in the main loop in my fork in the uart-fix branch of you want to get starting. That's working balancing code for a onewheel. 
The rest of the branch is not really compatible to the original FOC code, so you may only copy the balancing PID control.
I will eventually create a pull request, but I can't give you a date. I need to separate all the changes I did into independent branches before.

vfear, 2020-06-09:
> You can have a look in the main loop in my fork in the uart-fix branch of you want to get starting. That's working balancing code for a onewheel.
> The rest of the branch is not really compatible to [...]

did not quite understand what needs to be done?  =)

Chaos99, 2020-06-09:
This is the code in the main.c main loop that does the balancing.

`sensor_data.complete.Angle` is the angle of the board. I'm getting this from the sensor boards with stock firmware, which is not supported in the normal FOC branch. There is custom firmware for the sensor boards in @EmanuelFeru s Repo, but you need to figure out how to use it yourself and get the angle data from there.

So if you replace the code in `#ifdef VARIANT_HOVERBOARD` with the code below or create your own `#ifdef VARIANT_ONEWHEEL`, this might do the balancing for you.

If you are not comfortable with changing code yourself and doing a bit of debugging, you'll have to wait.
```
cmd1 = 0; // not needed
    if (sensor_data.complete.AA_55 == 85) // foot on sensor
    {      
      cmd2 = CLAMP(sensor_data.complete.Angle, INPUT_MIN, INPUT_MAX);  //angle data from sensor board
    }
    else
    {
      cmd2 = 0;
    }

//[...]

if (enable == 0 && (!errCode_Left && !errCode_Right) && (sensor_data.complete.AA_55 == 85))
  // inactive, no errors, foot on board
      {
        // only turn on for low angles
        if ( (cmd2 > -90 && cmd2 < 90))
        {
          shortBeep(6);                     // make 2 beeps indicating the motor enable
          HAL_Delay(20);                  // remove delay and 2nd beep if you need faster power on
          shortBeep(4); 
          //
          consoleLog("enabeling motors\r\n");
          enable = 1;                       // enable motors
        }        
      } 

//[...]

int8_t pOnE = 0; // 0=Proportional on Measurement (less overshoots); 1=Proportional on Error(textbook-PID)
      if (enable == 1) {
      if (ctrlModReq==2) { // for speed mode (doesn't work well)
         Kp = 0.035;  // critical oscillation  at 0.7
         Ki = 0.06;  // critical oscillation at 0.08
         Kd = 0.01;
      } 
      else // for voltage mode (works well)
      {
         Kp = 0.5; 
         Ki = 1.0;  
         Kd = 0.17; 
      }      

      int16_t input = -cmd2;
      int16_t error = cmd2;
      int16_t dinput = input - lastCmd;
      accel = Ki * error;

      if(!pOnE) accel-= Kp * dinput; 

      float output;
      if(pOnE) output = Kp * error; 
      else output = 0;

      output += accel - Kd * dinput;      
      speedL = output;
      speedR = speedL; // same speed for both motors, as they are connected in the wheel

      } else {       
        accel = 0;   // reset PID
        speedL = 0;
        speedR = speedL;

      }

if ( timeout < TIMEOUT) {
        #ifdef INVERT_R_DIRECTION
          pwmr = round(speedR);
        #else
          pwmr = round(-speedR);
        #endif
        #ifdef INVERT_L_DIRECTION
          pwml = round(-speedL);
        #else
          pwml = round(speedL);
        #endif
      }
 
    lastCmd = cmd2;   // save for next loop
    lastSpeed = speedL;
```

vfear, 2020-06-10:
yes, have to wait.  not very good at code.  then you need to make sure that the balance is turned on only when both legs are on the board.

vfear, 2020-07-08:
> That variant is in progress. The balancing controller is not yet implemented. It is planned to be implementer but I cannot give you a date. All the communication with sideboards is done, only the balancing stuff needs to be worked out.

is there any news on this topic?

vfear, 2020-07-10:
https://github.com/fungineers-youtube/DIY-Onewheel-Arduino-Code

Found this, can I somehow do onewheel mode, based on this?

EmanuelFeru, 2020-07-10:
Maybe is possible. I didn't look in detail. The challenging part for balancing for hoverboad is the middle joint, two angles.. It needs to be handled well.. For the moment it goes slow on the balancing controller due to other topics and holiday soon..

pigiit, 2020-07-11:
@EmanuelFeru:  Thanks for the great project!
I seems to me that at  the moment you only have communication with side boards for LEDs and Sensors, am I wrong?
If the code to read the angles is ready, could you add it to the project even if the motors control is missing?
 
Thanks

EmanuelFeru, 2020-07-12:
Also the angles from DMP is in place. Have a look at the structure received from the sideboards.

pigiit, 2020-07-12:
Thanks for the quick reply.
 I know the sideboards are sending the eulr angles (in may case only the L side one because I would like to use the serial control + left MPU  ) and I did something like that:

    // ####### SIDEBOARDS HANDLING #######
    #if defined(SIDEBOARD_SERIAL_USART2)
      sideboardLeds(&sideboard_leds_L);
      sideboardSensors((uint8_t)Sideboard_L.sensors);
      sideboardRoll((uint8_t)Sideboard_L.roll);
      sideboardYaw((uint8_t)Sideboard_L.yaw);
      sideboardPitch((uint8_t)Sideboard_L.pitch);

and for forw/back movement I just using the pitch value, is it correct?
Do you know the value variation I should have (-PI/2 + PI/2)?
Is a calibration needed?

Thanks

vfear, 2020-08-01:
have news on onewheel firmware?

vfear, 2020-09-02:
@Chaos99 @EmanuelFeru really nothing at all?

Chaos99, 2020-09-03:
@vfear Sorry. I wish it were different, but recently I had no time to work on the project. I need to fix a HW issue with the step-on sensors first before I even can test new firmware. 

My fork has the firmware I currently use in the "uart-fix" branch. It is good enough to ride the board (wear a helmet + protection!) but it's specific to my HW (STM32 Gen3-Board, Original-FW Sensorboards). 

The main control loop (and basically everything original to my fork) is in the main.c file.
Feel free to lift this out and place it into a clean copy of the official FOC within a new build target `VARIANT_ONEWHEEL`.
A simple merge will not do, as I removed most of the existing VARIANTS in main.c to help debugging.

If you are looking for the modifications for the STM32 Gen3 HW, you need to compare a lot of the setup and configuration files and headers. 

I hope I can pick up the slack later this year, but I won't make promises.

aghfars, 2021-01-31:
i flashed two side board and the main board too but motors are not moving.
all the mcu is GD32 and MPU6052C
What should I do?

EmanuelFeru, 2021-01-31:
The Variant_Hoverboard is not yet implemented.

aghfars, 2021-02-01:
so what can i do now
buy new Boards?

EmanuelFeru, 2021-02-02:
For the moment if you want original hoverboard functionality, that is the way to go. If you intend to use the hoverboard for other purposes than use this board that you just flashed.

vfear, 2021-03-26:
> @vfear Sorry. I wish it were different, but recently I had no time to work on the project. I need to fix a HW issue with the step-on sensors first before I even can test new firmware.
>
> My fork has [...]

Hello, any news?

gabyneo, 2021-04-26:
Hello there, I convert an hoverboard to a onewheel, but i have angle limite issue for the speed and confort. Someone have found a solution in original hoverboard controler firmware or pwm gyro by arduino ?

gabyneo, 2021-04-29:
Want hack hoverboard firmware to balancing with this arduino ( https://github.com/fungineers-youtube/DIY-Onewheel-Arduino-Code) from uart or I2C input instead the original gyro. Or hack the original gyro and implant kalman filter.

vfear, 2021-05-12:
> Want hack hoverboard firmware to balancing with this arduino ( https://github.com/fungineers-youtube/DIY-Onewheel-Arduino-Code) from uart or I2C input instead the original gyro. Or hack the original gyro and implant kalman filter.

https://youtu.be/9qGHqwf_8K8

https://forum.esk8.news/t/diy-onewheel-hoverboard/18862/209?u=vfear

vfear, 2021-05-12:
I'm not a programmer, so I haven't gotten a good balance job yet.  if there is news please let me know, I also do onewheel

Candas1, 2022-01-02:
Adding a ling to this [project](https://forum.esk8.news/t/yet-another-hoverboard-to-onewheel-project/60979/14) for visibility