EFeru/hoverboard-firmware-hack-FOC #63: DISABLE_LEFT/RIGHT ?

RoboDurden, 2020-06-12:
Not that important. But i have a lot of boards with 2 mosfet burned but one motor fully functional. The Fauth firmware still works but you z_errCode shuts everything down.
I tried to uncomment the two apperances in main.c but there is also error checking in the low level src files.

I am currently preparing an efficiency comparison for 250/350/450 Watt motors an would like to also compare with your firmware like i did in https://youtu.be/DILzw6RzPkE

Maybe you want to think about this issue for further updates.

EmanuelFeru, 2020-06-12:
Hey @RoboDurden . 

This a good suggestion, I will add it in the next commit/update. If you want to do it before my commit it is actually very simple, you just need to comment out the respective motor in bldc.c 

Comment out this if you don't need LEFT motor:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/558759b2c5873de84fbc5f6a5d0c324db34c8e22/Src/bldc.c#L173-L174

Comment out this if you don't need RIGHT motor:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/558759b2c5873de84fbc5f6a5d0c324db34c8e22/Src/bldc.c#L208-L209

There will be no error raised for the commented out motor, because the flag will stay at 0.

Also nice video! I gave you a comment regarding the 10" motors.

RoboDurden, 2020-06-12:
Fantastic !
Today i learned that the motor with 30mm magnets is the most powerful and most efficient. 
I will add your firmware to the test tomorrow.
Here my happy teaser: https://youtu.be/FasvKsVNvTA
Life is beautiful :-)

EmanuelFeru, 2020-06-13:
done.

RoboDurden, 2020-06-13:
Great ! 
I have just recorded the test drives with your firmware: it is about 23 % more energy efficient while beeing about 16% faster and on top way more silent.
Now i need to do some indoor recording to show the different magnet sides fo the different motors..

EmanuelFeru, 2020-06-13:
Awesome! It is very nice that you do this analysis.

RoboDurden, 2020-06-14:
Oh yes but i guess the efficiency would decrease. And as i do not live for the fun but the joy, i do not think i will make use of field weakening. P.S. in which file is FIELD_WEAK_MAX  used ?

My testdriving video is online: https://youtu.be/cYIpnW6RCSE
I dream you can enjoy that i did not really like your 'how to..'-video which isn't a full tutorial at all.

EmanuelFeru, 2020-06-14:
I have seen your video. Good benchmark video... Even though I find your comment at the beginning of the video really inappropriate. Making statements about someone you have never met in person, is already a big issue. If you don't like my project just say it and it's no problem, but do not extrapolate and say what I would think and feel because you don't have a clue.
 Aa and by the way I am Romanian not French.

Peace.

EmanuelFeru, 2020-06-14:
Topic can be closed.

closed 2020-06-14 by EmanuelFeru