EFeru/hoverboard-firmware-hack-FOC #64: Arduino v3.0 != 3.3V !

RoboDurden, 2020-06-14:
Hello again :-)
Now i to want bring your serial communication to my Arduino 3.3V Mini which should be no problem. 
I just notice `//  Arduino Nano 3.3V example code` and want to mention, that the Arduino Nano is running on 5V so the RX pin of the nano will output_pullup 5V to the stm32 pin which is said to be NOT 5V tolerant. 
So am i missing something ?

P.S. with my fork i defined the feedback struckt as

```
typedef struct{
   int16_t iSpeedL; // 100* km/h
   int16_t iSpeedR; // 100* km/h
   uint16_t iHallSkippedL;
   uint16_t iHallSkippedR;
   uint16_t iTemp;  // °C
   uint16_t iVolt;  // 100* V
   int16_t iAmpL;  // 100* A
   int16_t iAmpR;  // 100* A
   uint32_t crc;
} SerialFeedback;
SerialFeedback oFeedback;
```

I think the current is valuable information and the skipped steps could signal overpower.
Another nice value for feedback would be the total milage in meter.

Have not yet looked deeper in your field weakining but it seems that a level of weakining can be set. If so the send struct would need another byte. I would suggest one additional iCmdType byte and a 16 bit integer. Then you could implement future commands without the user have to update the arduino code !

the Roland and :-)

EmanuelFeru, 2020-06-16:
@RoboDurden you need to use the Right cable if you want to use Arduino. Only the Right cable is 5V tolerant! The Arduino operates at 5V indeed (I will update the text in the Arduino example).

See also below:
[image]

RoboDurden, 2020-06-16:
Thank you for that info. i always warn youtubers to not use 5V. But now i see that this is only correct for ADC whereas UART3 is okay with 5V :-)

i will open a uart issue today.. had to fork your nice firmware and implement my crc32 because your commuication yields very nasty steer -> -1000 commands at 9600 baud ...

joy :-)

EmanuelFeru, 2020-06-22:
Fixed.

closed 2020-06-22 by EmanuelFeru