EFeru/hoverboard-firmware-hack-FOC #66: Joystick control

farev, 2020-06-18:
Hello, this project was really helpful for me it is the only hoverboard-hack code that runs smoothly. I am building a mini go-kart and I want to control with an analog joystick I tried setting up the code for this but I couldn't, I can't control it very well. Can someone please help to set it up correctly for my analog joystick. 

Or is there a tutorial or instructions I can follow to correctly controll it with an arduino?

EmanuelFeru, 2020-06-18:
I'm happy to hear that you like the project. 
What type of joystick do you have? And I am a bit confused when I read Arduino and joystick together, what is the Arduino purpose?

As a side note you can also use 2 pedals with VARIANT_HOVERCAR of the firmware which is all already completely ready to use. You can check the Hovercar video in Readme page to see how it looks and if you like it.

farev, 2020-06-18:
I am using an analog joystick, that is just two potentiometers together. 
A picture of the joystick:
[https://www.google.com/search?q=analog+joystick&rlz=1C5CHFA_enGT877GT877&sxsrf=ALeKk01HagegYZcXhoeEO0exbg4JlSgH7Q:1592511648157&source=lnms&tbm=isch&sa=X&ved=2ahUKEwin0dqYmIzqAhVOn-AKHUIkDxEQ_AUoAXoECAwQAw&biw=1417&bih=671#imgrc=vSuR7hnQ3RTpAM](url)

When I mentioned to use an arduino I meant that if there are instructions I can follow to controll it with arduino instead of the joystick, because I do not know how to set it up correctly.

EmanuelFeru, 2020-06-18:
For joystick control is very easy:
- just select VARIANT_ADC in platformio.ini [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/85552b3e41f383475990a26f85ae4290f571388e/platformio.ini#L12-L20)
- then uncomment MID potentiomenters in config.h [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/85552b3e41f383475990a26f85ae4290f571388e/Inc/config.h#L227-L231)
- connect take the Left sensor cable (long wired) and connect your joystick, GNG to GND, one line to X joystick, the other line to Y joystick and take 3.3V from from the programming port. 
- Do not connect the red wire from the sensor board, it carries 15V! And it will burn the microcontroller. Also double check that the Black sensor wire is indeed GND, some boards have the wiring messed up.
- flash the firmware and give it a try
- at some point, you will have to calibrate your joysticks ADC readings to get it working properly, check config.h for more details

On the other hand, I still consider that pedals are more suitable for Go Karts. You can buy 2 of these for example:
https://www.banggood.com/custlink/mmG32e8i46

farev, 2020-06-18:
Than you, only how do I select VARIANT_ADC in platformio.ini, do I just remove the semi colon?

EmanuelFeru, 2020-06-18:
yes

farev, 2020-06-18:
I was able to get it all set up but it isn't working properly. It moves when it is at the center and when I move it forward or to the side it does not move the motors as it should.

EmanuelFeru, 2020-06-18:
When is at the center it moves because the ADC is not calibrated. And what do you mean by "not moving as it should"?

farev, 2020-06-18:
Likw when I try to move forwars only one motor moves forward, or when I try to turn it stops, it does different mivements than the ones to turn, or to move forward.
 How do I calibarate it? Maybe that is the problem.

EmanuelFeru, 2020-06-18:
Have a look here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/85552b3e41f383475990a26f85ae4290f571388e/Src/util.c#L398-L406

farev, 2020-06-18:
I did it but it is still the same.

farev, 2020-06-18:
Also when I turn it on it doesn't move until a little while after I start randomly moving the joystick.

farev, 2020-06-19:
I tried with a normal potentiometer and I calibarate it and I found out that both motors ,when trying only the speed, don't roatate at the same speed.

EmanuelFeru, 2020-06-19:
Can you try to swappin the motor wires? Sounds like the motor wires or hall sensors are not mapped correctly.
Also if you make a short video it will be easier to understand what is going on.

farev, 2020-06-19:
Here is a video:
[Video.zip](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/files/4805313/Video.zip)

EmanuelFeru, 2020-06-20:
From the video looks like the motor phases are not correct (even thought the colors are ok). Try swapping them. Apply small input and check how it behaves.

farev, 2020-06-20:
I tried sawapping the mtor wires and when it started moving it inmediatly turned off. I think my motherboard died because it doesn’t turn on any more.

farev, 2020-06-20:
It doesn't turn on with the button, I tried another button and nothing, but it does turn on if I connected to my computer with 3.3v connected. What may have happenned here?

EmanuelFeru, 2020-06-21:
I think one of the TIP27 might be damaged. Can you check it with an oscilloscope?

farev, 2020-06-21:
I got it to turn on again but the motors are still not moving correctly. I tried different phase mapping and it still dirrent work.

EmanuelFeru, 2020-06-23:
did you fixed it?

farev, 2020-06-23:
Yes, it turns on now but I still can't controll it correctly with the joystick. Do you know if someone else has tried to use an analog joystick?

farev, 2020-06-27:
Now the mainboard turns on and makes the sound but it then immediately turns off without making any sound. How do I check if a TIP27 is damaged with my multimeter? I tried it but couldn't figure out if something was wrong.

EmanuelFeru, 2020-06-28:
If it doesn't hold power on, then one TIP27 I guess is damaged. You can check them for continuity, I think one of them is shortened.

farev, 2020-06-28:
Some of them are passing like 0.2 or 0.7v and others are passing the complete voltage of the battery.

EmanuelFeru, 2020-06-28:
That I am not sure. Maybe you are better at electronics than me:
https://www.st.com/resource/en/datasheet/tip120.pdf

For me when they broke they were usually shortened.

joedeveloper, 2020-07-01:
Just FYI, I did a quick test with a cheap analog joystick connected to Arduino and then sending the massaged values to the hoverboard via uart, this works for me. Personally I find using direct connect adc joysticks potentially dangerous - I haven't tried on this firmware, so maybe the calibration / deadband / out of range logic is solid. The problem with wired joysticks are manyfold - from induction, to handling physical wear and tear a runaway condition can happen both on shorts and disconnects etc.

My plan is to have an esp32 connect to the hb over uart, then process control data from a wireless source - at least then you know you have to deal with out of bounds data gracefully.

farev, 2020-07-01:
Thank you for your help, could you send an example of your arduino code please. I have never used uart with an arduino.

joedeveloper, 2020-07-01:
I modified the arduino sample included in this repo only very slightly:
https://gist.github.com/joedeveloper/8b26ffa01f513a67e004164667ced5bc

the values I add / subtract here are specific to the cheap joystick I am using for testing.

EmanuelFeru, 2020-07-01:
I agree that joysticks resting at middle could be more dangerous. Normally, you should solder pull down resistors on the ADC pins, so in case of wire disconnection, the value is pulled down, making the ADC go out of range and by that it will go to a safety state.
For this protection the ADC_PROTECT_ENA needs to be activated in config.h

The classic potis resting at the lower bound (not middle) are safer to me.

Of course, depends on the application is better to work with digital signals. Like it was mentioned esp32, no need for ADC calibration, and the signal can be very clean, depending of the input source, for e.g. your smartphone or other digital device connected to the esp32.

farev, 2020-07-01:
Can I use a bluetooth hc05 or hc06 module the same way as a esp32?

EmanuelFeru, 2020-07-01:
I am not familiar with the bluethooth module, but I see they output UART data. What is the format of the UART data? Can it be programmed?
Because the mainboard expects a certain UART data format. ESP32 gives can be programmed so it is quite flexible, but I am not sure if the hc05 can be programmed or not.

farev, 2020-07-01:
No I would have to use an arduino to use it.

farev, 2020-07-01:
I am trying to control it directly from the arduino but I can't get it to move. Does the arduino need a seperate power source or can I connect it to the 3.3v in the mainboard?

EmanuelFeru, 2020-07-01:
For testing, you can power the Arduino externally. Use the example sketch.

farev, 2020-07-02:
I couldn't get the test working with the arduino, it just beeps.

EmanuelFeru, 2020-07-02:
What Variant are you using. Make sure you connected the arduino to the right cable and wires. Tx to Rx, Rx to Tx.

farev, 2020-07-02:
I am using the USART Variant and I connected it correctly, do I need to connect the gnd too?

EmanuelFeru, 2020-07-02:
Yes, sure. Otherwise there is no proper TX RX high and low level, it would be just floating input.

farev, 2020-07-02:
Even with the gnd connected the motors don't move when I try using the arduino.

EmanuelFeru, 2020-07-02:
That is strange.. in my testing it worked. Is there a posibility to check with an oscilloscope or an FTDI debuger to check if data is send from Arduino Tx, that data is send from Mainboard Tx?

farev, 2020-07-02:
Could this be happenning because of a faulty motherboard? I don't if my motherboard couldbe fualty.

EmanuelFeru, 2020-07-02:
If the melody is playing and the board turns on off properly, I think the board is ok.

farev, 2020-07-02:
I tried several times but I can't get it to work with the arduino, I tried checking with the arduino serial monitor if it was recieving and sending correctly but it didn't print what it should. The rx light in the arduino lights up but the tx doesn't, could the arduino be the problem?

EmanuelFeru, 2020-07-03:
On the Arduino did you use pin 2 and 3? Those are the pins that should go to the main board not Tx and Rx (see the Arduino code)

farev, 2020-07-03:
So I got the test running it worked fine, but when I tried it with the joystick and the modified arduino code it didn't work. It moved the wheels very little and not smoothly.

EmanuelFeru, 2020-07-03:
What was the issue?

EmanuelFeru, 2020-07-03:
Arduino code must be a bit optimized, that is just an example. Better is to do the sending in a periodic interrupt.

farev, 2020-07-03:
To get the test working I had to reflash again the motherboard becuase I found out that if you conected the pins wrong one time you had to flash it again to make it work. How do I do the sending in periodic interrupt?

EmanuelFeru, 2020-07-03:
https://www.instructables.com/id/Arduino-Timer-Interrupts/

farev, 2020-07-03:
I still couldn't get it to work with the arduino and the joystick even trying with periodic interrupt.

tmichele, 2020-07-05:
> For joystick control is very easy:
> 
> * just select VARIANT_ADC in platformio.ini [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/85552b3e41f383475990a26f85ae4290f571388e/platformio.ini#L12-L20)
> * then uncomment MID potentiomenters in config.h [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/85552b3e41f383475990a26f85ae4290f571388e/Inc/config.h#L227-L231)
> * take the Left sensor cable (long wired) and connect your joystick, GNG to GND, one line to X joystick, the other line to Y joystick and take 3.3V from from the programming port.
> * Do not connect the red wire from the sensor board, it carries 15V! And it will burn the microcontroller. Also double check that the Black sensor wire is indeed GND, some boards have the wiring messed up.
> * flash the firmware and give it a try
> * at some point, you will have to calibrate your joysticks ADC readings to get it working properly, check config.h for more details

Hi Emanuel, i followed every step:
- uncommented variant_adc at row 12 of platformio.ini
- uncommented row 226 and 230 of config.h
- flashed the motherboard with success (when i connect the battery it plays the melody)
Then i connected power and hall sensors to a motor and connected joystick to 3.3v (from programming port), gnd blue and green cable from left sensor.

When i move the joystick nothing happens.
What am i doing wrong? Is it ok to keep the power button jumpered all the time? If i connect the battery without jumpering the power pins it plays the melody and then it starts beeping

Thank you in advance.
[image]

Ps: every few minutes the motherboard beeps once.

EmanuelFeru, 2020-07-05:
Hi, it is not ok to keep button pressed all the time because it will get stuck in button pressed function. Press the button only during programing.

The beeping is most probably because the pots are not calibrated.

tmichele, 2020-07-05:
Ok, i flashed the motherboard while keeping pressed the power button (jumper), then i removed the jumper and disconnected the battery.
Then i've reconnected the original power button.

Now when i plug the battery the melody plays and then it start beeping (intermittent) even with no cable connected other than battery, is that ok?
While beeping i tried the _Auto-calibration of the ADC Limits_ but when i restart the board it plays the melody then it start intermittent beeps again.

EmanuelFeru, 2020-07-07:
Probably the calibration didn't succeed. Can you connect to serial debug and check the ADC values?

farev, 2020-07-08:
What values can I send from the arduino to make the motors move?

EmanuelFeru, 2020-07-08:
Initially you have to send 0 for the motors to enable, then you can send other values to spin them.

farev, 2020-07-08:
What is the range of the values? Is it from 0 to 255?

EmanuelFeru, 2020-07-08:
For the motors [-1000, 1000]

farev, 2020-07-10:
So, I got it working with the joystick but it only works while the arduino is connected to my computer.

farev, 2020-09-04:
Ok I got it working I changes the arduino code and now it works fine. It is very sensible, so for some applications it might be dangerous, but you can change the motors max speed very easily. There is a lot of commented code, this are some things I tried to make everything run even smoother but I left it there if someone wants to work on it. 

Link to my code: 
https://github.com/farev/Arduino-Code-for-Hoverboard-FOC-control)

[6 more replies]

closed 2020-09-04 by farev