EFeru/hoverboard-firmware-hack-FOC #67: Speed estimation

vanquang741987, 2020-06-19:
Hello EmanuelFeru
Your job about FOC is so great and amazing.
I am reading your code.
Do you have any document or general algorithm for speed estimation in your work?
I tried to read your code, but some times, I did not understand.
So, if you have, please give.
my email is vanquang741987@gmail.com
Thank you so much.

EmanuelFeru, 2020-06-19:
Good to hear that you like the project. 
The speed estimation is basically timing the duration between Hall changes. Have a look at the model webview on how is implemented,  and in docs\literature for the articles that I used.

vanquang741987, 2020-06-20:
Hello ~ EmanuelFeru
Thank you so much for your reply
Good to hear your answer.
     I tried to find and read  your document in docs\literature . But these document talk to sensorless algorithm, I could not find the docs for speed estimation using hall changes. 
    You known I can open your matlab model and their code. But sometimes, there are some conditions in speed estimation algorithm, So, I could not understand them. So, it is great if you provide documents or algorithm diagram for speed estimation. 
 Anyway, thank you so much for your time.
 Where are you come from?

EmanuelFeru, 2020-06-21:
For the moment all the documentation I have I put it on github. I am not sure if I will have time to write a more detailed documentation. Maybe in the future but I cannot guarantee.
I am from Romania :)

vanquang741987, 2020-06-21:
Hello ~ EmanuelFeru
Thank you for your reply
I hope I can get it.
In this time, I will try to read your algorithm
By the way, please take care your health by covid-19.
If you have document, please send to my email: vanquang741987@gmail.com
Best Regards.

closed 2020-06-23 by EmanuelFeru

m-farahmand, 2020-07-22:
Hi @EmanuelFeru and thanks for your efforts to making a good motor driving project.
I need to drive motor when it's not standstill and it has initial speed.
I have been used n_mot var to initial speed as work around(with mapping to speed boundary) but I wanna know Is there any good equation/solution to use motor speed in throttle?