EFeru/hoverboard-firmware-hack-FOC #68: your closed loop for SPEED mode :-/

RoboDurden, 2020-06-20:
I had to switch back to my firmware with that simple speed closed loop of my main.c .

Your closed loop (in BLDC_controller.c ?) is too tight to handle my sloppy chain drive. I now have spent some days to make my chain drive more perfect but the chains still get ripped of the sprockets.

Especially in curves. That may be because with steer=0 you send same speeds as pwml and pwmr to the motors. But in a curve the rpm of the motors never match.

In my closed lopp i calculate the target speed as the maximum of left and right. So both motors can 
drive with a stable pwm. In a right curve the right motor will draw less power at that pwm while the left motor is taking more load :-)

I don't see how to do this with your closed loop. From `pwml` i go to `rtU_Left.r_inpTgt` right into the low level `BLDC_controller_step()` of BLDC_controller.c . At that point there is no longer a left and right motor :-/

You might agree that an equal speed in a curve is a bug.
And i need to limit  my max speed to 6 kmh to be legal on German streets.

I also have the feeling that your closed loop is way to tight for me. It might be great with direct drive (motor = wheel) but i feel that when there is the slightest mismatch in speed the current imediately jumps up several amps.  Such load shocks then might rip apart my lose chain drive.

When i switched back to my old firmware where the pwm is reuglated quite slowly to match the actual speed i immedately could drive normaly with my solar car and no chains getting off.

So maybe you can give me a hint how to make your closed loop act more slowly. Like the low-pass filter in main.c

Also i liked my
```
#define MAX_RECUPERATION 3.0  //ROBO increase gas when more then 3.0 amps go back into the battery
  // my chain drive is to loose to take more then 3 Amps :-/
```

It is easier for my chains to get off while breaking. You too have `I_MOT_MAX` but i feel that only shuts the motors down. Maybe you can give me a tip how to add such a `MAX_RECUPERATION = 3` in your closed loop. In such a way that when the current goes under -3 , the speed will increase.

Greetings to Romania :-)

EmanuelFeru, 2020-06-23:
* Speed controller is highly dependent on the system inertia (which is a function of the Mass). So, normally re-tuning is needed to fit a specific application. That can be done by adjusting the `Kp` and `Ki` parameters:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/526cee4c0f71740839b42161d1902a97a8916765/Src/BLDC_controller_data.c#L298-L301
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/526cee4c0f71740839b42161d1902a97a8916765/Src/BLDC_controller_data.c#L325-L328
* and yes, the Left and Right motors can be set independently `rtU_Left.r_inpTgt` and `rtU_Right.r_inpTgt`, respectively

Thanks for the greetings ;)

RoboDurden, 2020-06-23:
But with steer=0 we don't know the speed of left and right wheel. We only know the speed the car should drive in a curve.
You could add a static speedMax and update it in the BLDC_controller_step() (i guess). Then when speed mode is active the closed loop would work on the max speed.
But i think the BLDC_controller.c is auto generated and it would not be a good idea to add manual changes.

closed 2020-07-20 by EmanuelFeru

RoboDurden, 2020-12-16:
Will it auto open when I post ?