EFeru/hoverboard-firmware-hack-FOC #69: Strange vibrations at +1000

RoboDurden, 2020-06-23:
Yesterday my motors vibrated heavily when pwm reached 1000. Either at full load uphill or when flat and target speed was out of reach. Then my closed loop also raises pwmlr to 1000.
It also happened downhill with fixed speed and up to 7 amps flowing back into the battery.
Now I have reduced the limits only in my variant uartCar to +-990 and the vibrations are gone.

Maybe you can guess immediately what happens at 1000 that doesn't happen at 990.

Two unlisted short videos for you:

https://youtu.be/19DJNsuxB_M
https://youtu.be/-fwnJrvL8Ug

Enjoy :-)

EmanuelFeru, 2020-06-23:
The thing is at 1000, there is high PWM applied to the motors.
* High PWM means the there is little window to measure Phase Currents, as a result, the Phase current measurements are not that accurate making the control vibrate a bit.
* I will lower the pwm_margin by 10

closed 2020-06-23 by EmanuelFeru

RoboDurden, 2020-06-23:
That vibrations did not show with my efficiency test drives. And even with my solar car the virbrations can not be felt by touching the motors. I gets noisy only because of the huge lightweight kelvar structure. So other users might not want you to reduce the max pwm. But i will be happy to test your new firmware in a few days :-)

EmanuelFeru, 2020-06-23:
The pwm reduction is by 1%, it won't be felt on the overall behavior.

commit 526cee4c (2020-06-23): Lowered pwm_margin by 10

- to avoid virbation at maximum PWM = 1000, due to reduced Phase current accuracy, the pwm margin is lowered by 10.
Fixed #69