EFeru/hoverboard-firmware-hack-FOC #70: FOC not free to use ?

RoboDurden, 2020-06-23:
I fear your GNU General Public License v3.0 license is no longer valid:

```
 * File: BLDC_controller.c
 * Code generated for Simulink model 'BLDC_controller'.
```

I only find license pricing for Matlab itself. But what about such auto generated code by Matlab ?

RoboDurden, 2020-06-23:
I only found this talk but i do not really understand it: https://au.mathworks.com/matlabcentral/answers/5988-does-mw-own-copyright-on-c-code-generated-by-matlab

Chaos99, 2020-06-23:
From what I read in that thread, allegedly copied from the licensing text of Matlab, you are free to use and distribute if it's part of a larger program (which in FOC I think is true, at least if all the STM libraries also count towards the program) and doesn't copy the functionality of Matlab itself (e.g. executing Matlab scripts), which FOC doesn't.

And as far as I know at least in my jurisdiction (Germany), converting something into another form doesn't change the copyright owner if it doesn't add artistic input. So if this is a C-version of a script written by Emanuel, it's still Emanuels copyright. (But if snippets of template code by Matlab are still part of it, of course they hold copyright on them too. But you are free to use, see above)

But i.a.n.a.l. of course.

RoboDurden, 2020-06-24:
I am not sure if the stm libraries are open source.
4. and 5. of http://www.st.com/software_license_agreement_liberty_v2

4. could mean that the code is not allowed to run on the GD32 clones which become more popular in these Chinese hoverboards.
5. I read as it is not allowed to turn the libs into open source. But I think I am wrong here and it is okay to let them be of open source as long as the stm files keep their original license.

If Emanuel holds a license for Mathlab he might offer his software for free.
But what about my fork ? I have no Mathlab license.

EmanuelFeru, 2020-06-24:
I would like to mention few things, maybe it will bring more clarity.

I developed the FOC controller completely from scratch, meaning that I didn't use any build in FOC library from Matlab. I only used summations, products, delays, min/max etc...basically, I made everything from scratch including PID controllers, even though Matlab has those as a library.  So, I hold the copyrights fully on the FOC implementation.

For generating code, I didn't use a dedicated STM library, or anything like that. So, no discussions of any copyrights implications with STM or other micro controller manufacturer because it is completely independent from that. Matlab is just generating plain C code, so there is no need to write it by hand. and of course I do have a license.

I use Matlab similar as a someone would use his garage tools and equipment to build "something". When he is done, he is the creator and holds the copyrights for that "something", not the manufacturer of his tools or equipment. So, I agree with @Chaos99 that I still have the copyright on the software, even if I brought into another form such as C code using my available tools. Now, because I give all this work for free, then is free. Use it and be happy with it :)
The difficulty comes when you want to modify it... then you'll need to buy the tools.

closed 2020-07-20 by EmanuelFeru