EFeru/hoverboard-firmware-hack-FOC #71: connection board

bekirbostanci, 2020-06-25:
Hi 
I did'nt connect with the hoverboard st link. I tried with connect stlink utilty but it was not work. 
What is the problem do you think?
[image]

EmanuelFeru, 2020-06-25:
Wrong pin connection maybe.. Check with a multimeter

KaanDemirkoparan, 2020-06-25:
Firstly he asked me about problem. And I checked everything (connections, stlink etc.) and they seems to be fine. He does every step right but it still gives that error. We couldn't solve it. Do you have any idea?

EmanuelFeru, 2020-06-25:
Have you tryied anothe ST link or another chip?

bekirbostanci, 2020-06-25:
yes i tried with stm32f103 (blue pill) and it worked

EmanuelFeru, 2020-06-25:
Have you tried with another main board too? Maybe the MCU is dead :(

bekirbostanci, 2020-06-25:
hoverboard is working, i controlled :(

EmanuelFeru, 2020-06-25:
That is strange

EmanuelFeru, 2020-06-27:
your mcu is really strange. I haven't seen such a thing before. I don't know what can be done more.

A yes, I know, cut your MCU out very carefully, clean the pads. Order new MCU and solder it on. You can find them on Aliexpress for quite cheap in batches. I did that already and I recommend.

bekirbostanci, 2020-06-27:
Yes, thank you for recomandation. I want to add information, I did not try but someone says that; boot0 pin should goes to and reset the card after the this process, you can erase.  
I dont now but it can work thaks again

closed 2020-06-27 by bekirbostanci