EFeru/hoverboard-firmware-hack-FOC #73: Does it work on all cards?

marco-teixeira, 2020-06-29:
Hello, does it work with any card? I have a LENZOD card, which looks a lot like yours, but when I plug in the st-link, the card keeps beeping, and is not recognized by the st-link program.

I already made the same comment on hoverboard-firmware-hack, I'll leave the link with photos of the board:
https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/152

EmanuelFeru, 2020-06-29:
I think someone had a similar issue with a red color mainboard
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues/71

Check my recommendation on the link above. I am not sure if you can something else than just   soldering another chip. I guess they locked the chip in a better way lately.

marco-teixeira, 2020-06-29:
Hello @EmanuelFeru , thank you so much for your answer. I believe I will buy another card, but now I am afraid of losing my investment. I am going to attach a link to a plaque sold in my country Brazil, could you tell me, with your experience, that it could work?
https://produto.mercadolivre.com.br/MLB-1339217555-placa-me-taotao-para-hoverboard-_JM?quantity=1#position=1&type=item&tracking_id=e443a25e-4718-4b05-a2b4-c63eb89f54ee

It says to be original taotao. I accept other tips, brands, which I must consider when purchasing the card. Probably the board that came with my hoverboard is parallel, I bought used.

EmanuelFeru, 2020-06-30:
This board should work, looks like is the original.

marco-teixeira, 2020-07-10:
Hello @EmanuelFeru , 

Sorry to bother you again, I thought well, and I'm going to buy an aliexpress card, which has already been validated by you. Could you please let me know the link to a card on aliexpress that you have used or are using?

Thankful,

EmanuelFeru, 2020-07-10:
@marco-teixeira I ordered from below. All is good, but on my second order, the wires are directly soldered on the board (without the white connectors) probably to cut the costs. I didn't really like that, but functionally is ok.
https://nl.aliexpress.com/item/4000384754004.html?spm=a2g0s.9042311.0.0.27424c4dnfNbJ0

adverst, 2020-10-02:
Hey I have this board too and initially had the same problem... But, pin order is different to the one in the images for the repo.
3.3v is the same as the board in the image, but then it is SWCLK, SWDIO, GND. If you get the pin order right this board flashes ok.

closed 2020-10-21 by EmanuelFeru