EFeru/hoverboard-firmware-hack-FOC #74: ADC limits calculation

BorMi, 2020-06-30:
I think that there is an error in cmd calculation for ADC variant, to allow going forward and backwards and right or left, i think that cmd's should be calculated in this way (Lines 710&717 of util.c):
        cmd1 = CLAMP(((adc_buffer.l_tx2 - ADC1_MIN_CAL) * (INPUT_MAX-INPUT_MIN) / (ADC1_MAX_CAL - ADC1_MIN_CAL))+INPUT_MIN, INPUT_MIN, INPUT_MAX);    // ADC1

        cmd2 = CLAMP(((adc_buffer.l_rx2 - ADC2_MIN_CAL) * (INPUT_MAX-INPUT_MIN) / (ADC2_MAX_CAL - ADC2_MIN_CAL))+INPUT_MIN, INPUT_MIN, INPUT_MAX);    // ADC2

(sorry, I do not have enough knowledge about git to send you a proper pull request)

EmanuelFeru, 2020-06-30:
Hi, thanks for the remark.
With a normal resting potis (meaning it rests at the minimum value), the cmd1, cmd2 can not go negative. That is why the minimum limit is 0. So, your equation is exactly the same as mine if you make `INPUT_MIN = 0`. I think for normal potis, it should stay like this, otherwise we need to fiddle in finding the middle to have no movement (`cmd1 = 0`, 'cmd2 = 0')
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/d8b529e063b1a03908cd1508e05af5b329eb348b/Src/util.c#L705-L711

---

If you want to have both backward and forward movement, the you have to use a mid resting potis. (Of course you can use a normal poti, but then you have to manually find the middle to have no movement). You can enable MID resting poti here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/d8b529e063b1a03908cd1508e05af5b329eb348b/Inc/config.h#L226
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/d8b529e063b1a03908cd1508e05af5b329eb348b/Inc/config.h#L230

---

Now, looking at the equation for MID resting poti, it is correct if `|INPUT_MIN| == INPUT_MAX`, but to be even more precise with respect to minimum, I will update it to this:
```
cmd1 = CLAMP((adc_buffer.l_tx2 - ADC1_MID_CAL) * INPUT_MAX / (ADC1_MAX_CAL - ADC1_MID_CAL), 0, INPUT_MAX)  
      +CLAMP((ADC1_MID_CAL - adc_buffer.l_tx2) * INPUT_MIN / (ADC1_MID_CAL - ADC1_MIN_CAL), INPUT_MIN, 0);    // ADC1 
```

BorMi, 2020-07-01:
Thanks for the explanation Emanuel,
I will enable the mid resting potis option, to be able to turn in both directions (I think that is not possible without the modification).
Many thanks for the amazing firmware!

closed 2020-07-20 by EmanuelFeru