EFeru/hoverboard-firmware-hack-FOC #77: Hovercar wheels turning without throttle pressed despite ADC calibration

thankthemaker, 2020-07-02:
Hey,
I used your hovercar variant, nice work.
But I have a problem with the calibration. I pressed the power button for ~5 seconds until the mainboard beeps. Then I pressed and released both pedals (potis) multiple times and waited for the timeout. 
When I turn on the mainboard, the wheels immediately start spinning at a slow speed. I use two of those pedals:

https://www.ebay.de/itm/Gaspedal-Steuerung-Kunststoff-Ersatz-Teile-Pedal-Universal-Schwarz-Montage/402266578924?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Another problem, when pressing the break pedal, sometimes the mainboard starts to beep and blink yellow. 

Thanks!

thankthemaker, 2020-07-02:
One more piece of information. I'm using the 3,3V from the programming header. The pedals normally use 5V.

EmanuelFeru, 2020-07-02:
Maybe the ADC calibration was not fully successful.
The ADC values should be above 300 and below 3795 for the calibration to be accepted.
You can also try to un-comment this line and try again:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/22984a7fd60b164609481768f952b8aa529c6f3d/Inc/config.h#L358

However, to fully understand the issue, best is to check the ADC values using the Serial debug. Because I don't know what to expect from the pedals (they are new to me). You can do that by flashing VARIANT_ADC.

And are you using a sideboard connected to mainboard?

thankthemaker, 2020-07-02:
Thanks for your reply! Yes, I use a sideboard. 

Ok, I try to comment SIDEBOARD_SERIAL_USART3 and  FEEDBACK_SERIAL_USART3 and enable DEBUG_SERIAL_USART3 to see the ADC values or do I really need to flash VARIANT_ADC for that?

I also will try to disable the ADC protection temporary and check if it fixes the error with the brake pedal.

BTW: How did you connect the LEDs to get both sides working? Did you solder both sides to the same wire and connect them to a single port of the sideboard?

EmanuelFeru, 2020-07-02:
Both can work, keep Variant_Hovercar with Debug or Variant ADC.

To get both sides LEDs, I connected them in parrallel to LED4 and LED5 (to get upper and lower side in control). I used the [GD sideboard](https://github.com/EmanuelFeru/hoverboard-sideboard-hack-GD) The front LEDs I have them separate, on the AUX3 pin. In parallel with the blue LEDs didn't work (the blue ones won't light up any more if in parallel with front LEDs).

thankthemaker, 2020-07-02:
Cool, thanks for the info. I already guessed you connected them in parallel but I was not sure.

Ok, flashed the VARIENT_ADC and attached serial monitor to UART. Here's what i measured.

`-- Motors enabled --
1:1114 2:1768 3:296 4:296 5:1461 6:4005 7:1747 8:179
1:1113 2:1553 3:258 4:258 5:1463 6:4005 7:1734 8:204
1:1113 2:1857 3:262 4:262 5:1463 6:4005 7:1724 8:224
1:1113 2:1537 3:290 4:290 5:1461 6:4005 7:1716 8:239
1:1114 2:1699 3:268 4:268 5:1462 6:4008 7:1710 8:251
1:1115 2:1625 3:282 4:282 5:1461 6:4008 7:1705 8:261
1:1113 2:1639 3:287 4:287 5:1461 6:4008 7:1702 8:267
1:1113 2:1633 3:255 4:255 5:1463 6:4008 7:1698 8:274
1:1113 2:1649 3:274 4:274 5:1461 6:4008 7:1696 8:278
1:1113 2:1805 3:253 4:253 5:1463 6:4008 7:1694 8:282
1:1113 2:1117 3:285 4:285 5:1461 6:4010 7:1693 8:284
1:1113 2:2537 3:239 4:239 5:1463 6:4010 7:1691 8:288
1:1114 2:2679 3:253 4:253 5:1463 6:4010 7:1690 8:290
1:1113 2:1333 3:286 4:286 5:1459 6:4010 7:1690 8:290
1:1113 2:1673 3:269 4:269 5:1461 6:4010 7:1689 8:292
1:1113 2:1118 3:305 4:305 5:1456 6:4010 7:1688 8:294
1:1113 2:1145 3:268 4:268 5:1463 6:4010 7:1688 8:294
1:1113 2:1711 3:271 4:271 5:1462 6:4010 7:1688 8:294
1:1113 2:1743 3:262 4:262 5:1463 6:4013 7:1687 8:296
1:1113 2:1617 3:243 4:243 5:1465 6:4013 7:1687 8:296
1:1113 2:1885 3:248 4:248 5:1463 6:4013 7:1687 8:296
1:1114 2:2640 3:281 4:281 5:1461 6:4013 7:1687 8:296
1:1113 2:1517 3:270 4:270 5:1461 6:4013 7:1686 8:298
1:1112 2:2235 3:255 4:255 5:1463 6:4013 7:1686 8:298
1:1113 2:1141 3:237 4:237 5:1463 6:4013 7:1686 8:298
1:1113 2:1113 3:253 4:253 5:1462 6:4013 7:1686 8:298
1:1114 2:2117 3:268 4:268 5:1463 6:4013 7:1685 8:300
1:1113 2:1979 3:268 4:268 5:1463 6:4016 7:1685 8:300
1:1114 2:1689 3:258 4:258 5:1463 6:4016 7:1685 8:300
1:1113 2:1355 3:286 4:286 5:1461 6:4016 7:1685 8:300
1:1113 2:1765 3:280 4:280 5:1461 6:4016 7:1685 8:300
1:1114 2:1665 3:271 4:271 5:1462 6:4016 7:1685 8:300
1:1113 2:1523 3:266 4:266 5:1463 6:4016 7:1684 8:302
1:1113 2:1901 3:273 4:273 5:1463 6:4016 7:1684 8:302
1:1113 2:1715 3:263 4:263 5:1463 6:4016 7:1684 8:302
1:1113 2:1682 3:258 4:258 5:1463 6:4016 7:1684 8:302
1:1114 2:2137 3:280 4:280 5:1463 6:4016 7:1684 8:302
1:1113 2:1772 3:269 4:269 5:1461 6:4016 7:1684 8:302
1:1113 2:2531 3:254 4:254 5:1463 6:4016 7:1684 8:302
1:1113 2:2461 3:260 4:260 5:1463 6:4016 7:1684 8:302
1:1113 2:1139 3:283 4:283 5:1459 6:4016 7:1684 8:302
1:1113 2:1777 3:269 4:269 5:1461 6:4019 7:1684 8:302
1:1112 2:1607 3:279 4:279 5:1461 6:4019 7:1684 8:302
1:1113 2:1869 3:269 4:269 5:1463 6:4019 7:1684 8:302
1:1113 2:2007 3:293 4:293 5:1459 6:4019 7:1683 8:304
1:1113 2:1785 3:357 4:357 5:1461 6:4019 7:1683 8:304
1:1113 2:1945 3:431 4:431 5:1456 6:4019 7:1683 8:304
1:1113 2:2155 3:514 4:514 5:1443 6:4019 7:1683 8:304
1:1113 2:3693 3:583 4:583 5:1453 6:4016 7:1683 8:304
1:1113 2:2665 3:719 4:719 5:1443 6:4016 7:1683 8:304
1:1110 2:3015 3:904 4:904 5:1434 6:4016 7:1683 8:304
1:1113 2:3177 3:955 4:1000 5:1449 6:4013 7:1683 8:304
1:1455 2:3197 3:933 4:1000 5:1457 6:4013 7:1683 8:304
1:1441 2:3536 3:928 4:1000 5:1457 6:4013 7:1683 8:304
1:1695 2:3737 3:931 4:1000 5:1455 6:4013 7:1682 8:306
1:1112 2:3203 3:930 4:1000 5:1455 6:4013 7:1682 8:306
1:1113 2:3203 3:925 4:1000 5:1456 6:4013 7:1682 8:306
1:1772 2:3203 3:922 4:1000 5:1452 6:4013 7:1682 8:306
1:1651 2:1752 3:784 4:925 5:1469 6:4013 7:1682 8:306
1:1113 2:1525 3:273 4:285 5:1493 6:4016 7:1682 8:306
1:1113 2:1499 3:297 4:297 5:1459 6:4016 7:1682 8:306
1:1113 2:1733 3:265 4:265 5:1463 6:4016 7:1682 8:306
1:1113 2:1119 3:289 4:289 5:1457 6:4016 7:1682 8:306
1:1112 2:2603 3:268 4:268 5:1461 6:4016 7:1682 8:306
1:1113 2:1737 3:292 4:292 5:1459 6:4016 7:1682 8:306
[... 22 lines cut ...]
1:1115 2:1545 3:261 4:261 5:1461 6:4019 7:1698 8:274
1:1113 2:1650 3:252 4:252 5:1463 6:4019 7:1694 8:282
1:1113 2:1733 3:265 4:265 5:1463 6:4019 7:1691 8:288
1:1113 2:1951 3:269 4:269 5:1462 6:4019 7:1689 8:292
1:1115 2:1131 3:270 4:270 5:1461 6:4019 7:1687 8:296
1:1114 2:2545 3:260 4:260 5:1463 6:4019 7:1686 8:298
1:1113 2:1147 3:249 4:249 5:1461 6:4019 7:1684 8:302
1:1113 2:2672 3:274 4:274 5:1461 6:4019 7:1684 8:302
1:1113 2:1689 3:279 4:279 5:1460 6:4019 7:1683 8:304
1:1113 2:1795 3:302 4:302 5:1457 6:4019 7:1682 8:306
1:1113 2:1680 3:267 4:267 5:1463 6:4019 7:1682 8:306
1:1113 2:2045 3:289 4:289 5:1457 6:4019 7:1682 8:306
1:1114 2:1161 3:265 4:265 5:1461 6:4019 7:1681 8:308
1:1113 2:1707 3:272 4:272 5:1459 6:4019 7:1681 8:308
1:1113 2:1573 3:258 4:258 5:1461 6:4019 7:1681 8:308
1:1113 2:1763 3:288 4:288 5:1459 6:4019 7:1681 8:308
1:1113 2:2525 3:254 4:254 5:1462 6:4019 7:1681 8:308
1:1113 2:1163 3:264 4:264 5:1461 6:4019 7:1680 8:310
1:1113 2:1763 3:256 4:256 5:1461 6:4019 7:1680 8:310
1:1113 2:2005 3:270 4:270 5:1461 6:4019 7:1680 8:310
1:1113 2:2275 3:251 4:251 5:1461 6:4019 7:1680 8:310
1:1113 2:1736 3:268 4:268 5:1461 6:4019 7:1680 8:310
1:1112 2:1643 3:290 4:290 5:1459 6:4019 7:1680 8:310
1:1113 2:1705 3:278 4:278 5:1459 6:4019 7:168
--- exit ---`

EmanuelFeru, 2020-07-02:
Is there a reason why `3:...` and `4:...` are not 0 when the pedals are released?. Is this after or before the calibration?

thankthemaker, 2020-07-02:
I have no idea why the are not 0. That seems to be my problem why the wheels are spinning.
I did the calibration several times, but I'm not sure if it was successful, so it's after calibration.

Meanwhile I commented out the ADC protection and it solves the problem with the error (beep and yellow led).

thankthemaker, 2020-07-02:
In the config.h there is a ADC1_MIN of 1000. If I understand it correctly, my pedal "sends" 1114 when released. So I guess the calibration failed and the so the firmware takes the difference of 114 (1114-1000) as the input throttle, right?

I'll try a modified firmware with ADC1_MIN=1115 and ADC1_MAX=3500. These seems the range of the throttle pedal. 

The brake pedal looks a little strange to me. It seems to oscillate even when released. When the throttle pedal was pressed, the brake pedal was released.

EmanuelFeru, 2020-07-02:
Yes the ADC MIN you have to change. For me it was 1000. Same procedure for ADC2.
I would be curious why the calibration failed.

If the wheels oscillate when pressing the brake and throttle, that is OK. Because the torque mode needs to operate properly. So with wheels on the ground it will work correctly.
For testing with wheels off the ground use VOltage mode. Then for smoothnees on the ground you can switch back to torque mode.

thankthemaker, 2020-07-02:
Ok, is it normal that in debug value 2 is throttle? I found out, that val 1 is my brake pedal and val 2 is throttle. So throttle is already oscillating when both pedals are released.

thankthemaker, 2020-07-02:
I once again tried to calibrate, looks good in DEBUG but has no effect. Also changing the values in config.h doesn't seem to have any effect.

```
1:1115 2:1871 3:285 4:285 5:1624 6:4473 7:1658 8:353
1:1115 2:2369 3:302 4:302 5:1625 6:4473 7:1658 8:353
1:1115 2:2456 3:294 4:294 5:1625 6:4473 7:1658 8:353
-- Motors disabled --
-- Motors enabled --
ADC calibration started... OK
-- Motors enabled --
1:1115 2:1441 3:343 4:343 5:1610 6:4476 7:1804 8:67
1:1115 2:2028 3:308 4:308 5:1624 6:4476 7:1771 8:132
1:1114 2:1843 3:317 4:317 5:1624 6:4476 7:1745 8:183
1:1114 2:1988 3:314 4:314 5:1623 6:4476 7:1725 8:222
1:1115 2:1963 3:304 4:304 5:1624 6:4473 7:1709 8:253
```

```
  //#define ADC_PROTECT_ENA               // ADC Protection Enable flag. Use this flag to make sure the ADC is protected when GND or Vcc wire is disconnected
  #define ADC_PROTECT_TIMEOUT 100       // ADC Protection: number of wrong / missing input commands before safety state is taken
  #define ADC_PROTECT_THRESH  300       // ADC Protection threshold below/above the MIN/MAX ADC values
  #define ADC1_MIN            1115      // min ADC1-value while poti at minimum-position (0 - 4095)
  #define ADC1_MAX            3500      // max ADC1-value while poti at maximum-position (0 - 4095)
  #define ADC2_MIN            2000       // min ADC2-value while poti at minimum-position (0 - 4095)
  #define ADC2_MAX            3500      // max ADC2-value while poti at maximum-position (0 - 4095)
  ```

thankthemaker, 2020-07-02:
I checked the pedals with an ESP8266. It uses 3.3V with 10bit, so the range is "only" 0-1023
Here are my results:

ESP8266 A0 "break pedal"
297 - 842
ESP8266 A0 "throttle pedal"
297 - 825

Both pedals work as expected on an ESP8266.

EmanuelFeru, 2020-07-02:
> I once again tried to calibrate, looks good in DEBUG but has no effect. Also changing the values in config.h doesn't seem to have any effect.

Please change the FLASH_WRITE_KEY to another number, otherwise it will use the values from calibration (possibly wrong values):
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/22984a7fd60b164609481768f952b8aa529c6f3d/Inc/config.h#L155

One more commnent, when you calibrate ADC, for the `MIN = mimimum_found +100`, and for the `MAX = maximum_found -100`, to make sure you get a nice `0` when is at MIN and a nice `max` when you are at MAX.

thankthemaker, 2020-07-03:
Hey,
I have no idea why the values for brake `(1:...)` are good and for throttle` (2:...)` are jumping so much. The pedals are ok, you can see this in my test with an ESP8266, both pedals deliver almost the same results using them with 3,3V on ESP.
 
But using them with the STM32 in the hoverboard mainboard I got different results for both pedals. 
I'll check the cables from the mainboard to the JST-connector, maybe a cable isn't ok/broken?

EmanuelFeru, 2020-07-03:
What happens if you swap the pedals?

thankthemaker, 2020-07-04:
It makes no difference. I guess there is something wrong with the ADC on the motherboard. ADC1 looks good, but ADC2 now always, even if pedal is completely released, shows values >3000. Maybe ADC2 is broken.

I'll try to get some simple potentiometers and look if they are behaving different.
Otherwise I'll have to switch to UART and use the analog input from the ESP. I already did this before I decided to use the hovercar variant.

thankthemaker, 2020-07-05:
Hmm, I tried different things and also fried my mainboard. I changed the TIP127 and also tried another board. Always the same result. I now switched to UART and I'm going to modify your firmware to also have a mix of UART and HOVERCAR. 
Anyway I wanted to use an ESP8622 for controlling lights and the speed settings.

So, for now I'll close this ticket. If I have any new information, I'll update the issue.

closed 2020-07-05 by thankthemaker

Candas1, 2020-08-08:
Hi,

Glad I found this issue.
I am actually experiencing the same with a hall sensor based ebike throttle. 
I need only one input so I just connected both adc1 and adc2 together, but adc2 is changing a lot, it's weird. 

I will do more debug and share.
I can at least switch adc input for speed and steering to see if it works better.

thankthemaker, 2020-08-08:
> I am actually experiencing the same with a hall sensor based ebike throttle.

I have to syy, mine are also Hall-sensor based pedals. Maybe it makes a difference to use Hall-sensors instead of potis.

Candas1, 2020-08-08:
I wonder why adc2 has that sampling time, it seems to work better if I comment that line of code
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/b1f7c4c296e3123d64000290f9ef629318bcb562/Src/setup.c#L697

But I think it works with your hovercar without a problem Emanuel