EFeru/hoverboard-firmware-hack-FOC #78: UART with Raspberry-pi or Jetson-nano

Branch-h, 2020-07-06:
Hi,
@EmanuelFeru  could you add any python or c++ sample code for sending speed and steer (controlling) hoverboard using Raspberry-pi or Jetson nano, please. I tried to conect with UART using python and pyserial to send speed and steer, but without any success. I don't understand how to package data to send it with pyserial, so any info will be helpful.
Any help would be really appreciated.

EmanuelFeru, 2020-07-06:
Hi,
I am not very familiar with python, so I cannot provide an example. However, the idea is the same, you have to pack your data into a structure of this form:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/22984a7fd60b164609481768f952b8aa529c6f3d/02_Arduino/hoverserial/hoverserial.ino#L43-L49

Pack the data and then send it on UART all together (on Arduino is like this):
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/22984a7fd60b164609481768f952b8aa529c6f3d/02_Arduino/hoverserial/hoverserial.ino#L76-L86

Call the `Send(.., ..)`periodically. Make sure that there is some time gaps between the sends,  because the mainboards Rx relies on UART idle line detection, to know that a full package has arrived and is time to generate an interrupt to process the Rx data.

Branch-h, 2020-07-08:
@EmanuelFeru  thanks for help, I finally made it.

Here is an example if anyone need it:
It's probably not the best but it works.

[uart_python.txt](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/files/4890854/uart_python.txt)

closed 2020-07-08 by Branch-h

darld, 2021-07-20:
@Branch-h I am using your python script, but the wheel have not response until the end of the script (550 steps). 
I am using jetson nano b01 4gb and all the setting is follow the @EmanuelFeru  repo

Branch-h, 2021-07-22:
@darld I used that script on orangepi-zero, you should figure out to somehow set UART (port)  on jetson nano to make it work. Maybe there was some changes in @EmanuelFeru repo so commands don't work the same.