EFeru/hoverboard-firmware-hack-FOC #79: GD32F103RCT6 flashing

trondin, 2020-07-08:
Hi,
is any difference for GD32F103RCT6 flashing? Or I have to use it as  STM32F103RCT6 in PlatformIO?

EmanuelFeru, 2020-07-08:
As far as i know both can be flashed the same with no additional change. GD32 seems to be a clone of the STM32.

trondin, 2020-07-09:
Thanks, it really works without any problem. The board had survived :)

closed 2020-07-09 by trondin

bekirbostanci, 2020-07-28:
i have same mcu and flashing give this alert "Read out protection is activated".
I tried Target > Option Bytes > Read Out Protection > Enable than apply but 
16:10:06 : Option bytes updated successfully.
16:10:06 : Can not read memory!
                 Disable Read Out Protection and retry.

Do you have any idea about this problem ?

trondin, 2020-07-28:
Are you sure your system powered from battery, not from programmer - I have killed one processor by powering from programmer. After processor replacement the main board is functional again - but it was different board, STM32 based. For GD32F103RCT6 protection was remover without any problem.

bekirbostanci, 2020-07-28:
@trondin I tried again and solved problem thank you