EFeru/hoverboard-firmware-hack-FOC #80: No motor movement - any suggestions

djmrozinski, 2020-07-15:
Hello,

I have a esp32 providing a 0v to 3v input to the blue wire of the left UART connector.  I have the steering parameter set to zero, no steering and I selected the ADC variant in the config.h  No connection to the steering input on the left UART.

The firmware compiles and the board flashes fine.  When the start button is reconnected after flashing I get the start up tones.  I have confirmed that the ESP32 is outputting a voltage to the UART port. 

I feel that I am missing something here.  Do I need to make the ADC variant selection in the platformio file as well.  Looks like either should work.

I have used this same setup with the bipropellant firmware hack and it works.

Been messing with this for a few hours with no success.  Any suggestions.

Thanks
djmrozinski

EmanuelFeru, 2020-07-15:
If you use platformio, I suggest uncomenting desired variant there. How are you sending an analog voltage from esp32? Using DAC?
Note that for ADC variant you need to calibrate the ADC input ranges. 
For debugging you can connect to the right cable to see if the values are correct.

When turning the board on, the ADC input needs to be very close to 0, to enable the motors(for safety).

FirePT, 2020-07-15:
Hi @djmrozinski, are you trying to utilize the two available DAC channels that are explained here : https://www.youtube.com/watch?v=T4jg1j0XgWU  ??

That looks to be a great idea and I was looking into serial UART but this could be the best option in fact.

Can you take a look @EmanuelFeru and give your feedback ?

EmanuelFeru, 2020-07-15:
@FirePT using the 2 DAC should work. The resolution is not that high, only 255 steps, but maybe is sufficient for this purpose. 
So, I don't see a reason why it shouldn't be OK and also easy to implement.

The UART on the other hand has the advantage of full resolution 1000 steps. And can be augmented will any functionality you can imagine..buttons, mode changes, bi directional communication (feedback from the mainboardl) etc.

djmrozinski, 2020-07-15:
Couple of items.  
When I power on the board, I hear the normal startup tone then there is on short additional tone.  Does that mean anything??

I check the output from the DAC of the ESP32 at zero and I am getting about .14 volts on the output, is that close enough to zero for start up.  

Also can you point me to any resource that would explain the calibration process.

Thanks for the help
djmrozinski

EmanuelFeru, 2020-07-15:
The additional tone is the signal that the motors are enabled. So, theoretically it should listen to commands.

But looking at the wiring, the throttle should be connected to ADC2, so the green wire (not the blue), blue is steering.

For calibration, check config.h, also I implemented an auto-calibration, search for auto-calibration in until.c

FirePT, 2020-07-30:
Hi, @EmanuelFeru, I'm opening this same problem because I received some material and tried to implement the ADC and getting the exactly same result, just like @djmrozinski. I have checked everything with a multimeter and the voltagem outputs are fine but nothing happens on the motors. I've tried with the ESP32 DAC and because that failed, tried with a poti. I got the same result... no movement.. none.

a) I got he firmware from:

https://pionierland.de/hoverhack/ , selected this fork

b) uncommented the line :

#define VARIANT_ADC

c) flashed without problems (GD32 variant board)

On button pressed I hear the init sound with the upper scaling but not the second bip @djmrozinski refers to.

d) using either ESP32 DAC or poti to provide a measured 0v up to ~2v on GREEN cable

nothing happens...

If there is any error with the firmware does it get printed on the serial connection on RIGHT CABLE ?
What software do you use to monitor this serial ?
What UART-2-USB you use ? Can I use StLink ?

Thanks for your time in advance :)

djmrozinski, 2020-07-30:
What corrected my problem Ws connecting  sure both ADC channel. I wasn’t using steering so I didn’t connect the steering channel.  Once connected I got the enabled tone an all worked well. 
Dave

FirePT, 2020-07-31:
@djmrozinski thanks! I was doing the exactly same thing :) . Unfortunately I realized that hoverboard motors doesn't fit my requirements... have to go up in rpm's. I'll have to think on a way to still use this in another project :)

closed 2020-10-21 by EmanuelFeru