EFeru/hoverboard-firmware-hack-FOC #81: How to change Torque Target in Torque Mode to achieve electronic braking?

fender90, 2020-07-16:
Hi Emanuel.
First let me congratulate you on your fantastic job!
I am making a robotic platform which should be controlled via Arduino. At the moment I have connected a simple RC receiver to it but I plan to implement a more sophisticated ground control unit later. 
I have selected torque mode in the firmware and would like to have something like electronic braking.
The documentation says the following.
_TORQUE MODE: in this mode the input torque target is realized. This mode enables motor "freewheeling" when the torque target is 0. Recommended for most applications with a sitting human driver. If motor braking is desired instead of "freewheel" when torque target is 0, then a torque target below 0 should be set when speedAvgAbs > 0._

Could you please specify on how to do that? I didn't find where I can change the torque target value. Also is it possible to have some kind of "stationary brake" option? Maybe when speedAvgAbs >= 0._ ?

juodumas, 2020-07-16:
@EmanuelFeru looks like this question comes a lot and everyone seems to be reading the README. So I would like to suggest to extend the torque mode description and clarify its differences/advantages/disadvantages to other modes...

fender90, 2020-07-16:
That would be perfect. Thank you very much.

EmanuelFeru, 2020-07-16:
That can be implemented, if i have time I will do it over the weekend.

EmanuelFeru, 2020-07-18:
@fender90 question, why have you decided to use Torque mode and not Voltage or Speed for your robot?
Because Voltage or Speed will stop when input is 0.

fender90, 2020-07-18:
> @fender90 question, why have you decided to use Torque mode and not Voltage or Speed for your robot?
> Because Voltage or Speed will stop when input is 0.

Because the platform I am making is going to carry heavy load.

EmanuelFeru, 2020-07-18:
With Voltage or Speed you can carry the same load. The control mode does not influence the maximum load it can carry.

fender90, 2020-07-18:
> With Voltage or Speed you can carry the same load. The control mode does not influence the maximum load it can carry.

O_o Thanks for the clarification, but then it is not clear what is the purpose of other modes and what are the advantages/disadvantages of each one.

EmanuelFeru, 2020-07-19:
@fender90, @juodumas  In my last commit f2d86f3b309b722ae705d0149b7557732cc0c202 I added Electric Brake and Standstill hold functionality. Also I added a table in Readme to help you choose the methods that suits your application and needs.
[image]

Feel free try and choose the behavior that you like or need.

fender90, 2020-07-20:
@EmanuelFeru thanks for the updates. Will close the issue.

closed 2020-07-20 by fender90