EFeru/hoverboard-firmware-hack-FOC #82: split 2 mainboards version

happytm, 2020-07-16:
I just bought new hoverboard linked below:

 [https://www.ebay.com/itm/Bluetooth-Hoverboard-Self-Balancing-Electric-Scooter-LED-UL2272-CE-no-Bag-Sports/353089292564?ssPageName=STRK%3AMEBIDX%3AIT&var=622283901145&_trksid=p2057872.m2749.l2649](url)

It came with 2 split mainboards with each having their own micro controllers

I searched for the schematic for these boards and was able to find schematic attached.

[2 Mainboards_hoverboard_schematic.pdf](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/files/4934354/2.Mainboards_hoverboard_schematic.pdf)

I want to use this hoverboard as a robotic platform using wifi control from smartphone. What is the best way to control it either your firmware or existing firmware ?

Thanks.

happytm, 2020-07-17:
I would like to use ESP32 micro controller to control the motor direction and steering. I have front end code for ESP32 which is a virtual  joystick sending X & Y coordinates. Balancing part is not important for this application. Basically I want to make powerful toy car out of it.

Thanks.

EmanuelFeru, 2020-07-20:
> What is the best way to control it either with your firmware or existing firmware ?

You have several options, ADC, UART, PWM, PPM. Control via ESP32, I would choose UART control to the mainboard, but other should work well too.

happytm, 2020-07-25:
Looked like sideboards above use following circuit described below:

https://aidilj.com/2015/01/sensored-bldc-motor-control-with-arduino/

Thanks.

closed 2020-10-21 by EmanuelFeru