EFeru/hoverboard-firmware-hack-FOC #86: Help on my project : need more rpm/speed

FirePT, 2020-08-03:
Hi @EmanuelFeru , since I now have everything working i realized that SPEED mode is not enough for my project (a tennis ball throwing machine). Looking into the firmware I see all the warnings around the field weakening LO/HI values and I think you have that warnings because of the normal usage in "vehicle" mode but in my project I will not use it to locomotion but only to drive the ball forward. Can you help me try to archive more rpm from the motors ?

I already changed the default FIELD_WEAK_MAX  from 5 to 8 but I think it didn't made any change...

Thanks!

EmanuelFeru, 2020-08-03:
If you just need fast speed, then I recommend Sinusoidal control type with Phase advance, both selectable in config.h.

FirePT, 2020-08-03:
Thanks! I'll try it.

FirePT, 2020-08-06:
@EmanuelFeru thank you! It worked, I'm still tweaking the settings but the difference is enormous in rpm.
Currently I have :

// Control selections
#define CTRL_TYP_SEL    SIN_CTRL
#define CTRL_MOD_REQ    VLT_MODE

// Field Weakening / Phase Advance
#define FIELD_WEAK_ENA  1
#define PHASE_ADV_MAX   45    //from default 25

I have one question regarding :

#define N_MOT_MAX       1000            // [rpm] Maximum motor speed limit

do you think it's safe to increase this value in terms of integrity of the motors/board ?
With the settings I defined above I already concerned by the amount of heat and how quickly it rises in both motor, board and battery... I don't think for now that I can maintain it for more than 1 minute safely. I'm testing a start and stop ball sensor so it just maxes velocity an instant before it have to throw the ball and slows down until a new one triggers the sensor again.

Thanks for your help (and patience) !

benjaf, 2020-08-06:
@FirePT If you need more speed without using excessive amounts of power, you might consider modifying the motor to a 'delta' configuration (see issue #26). This trades some of the torque for speed, which seems like a good deal in this case.

EmanuelFeru, 2020-08-06:
@FirePT
> do you think it's safe to increase this value in terms of integrity of the motors/board ?

You can increase that number to 1500 to avoid fixed point overflows. Personally, the maximum I went was around 1200 rpm. It was for a short period, but it felt a quite dangerous, and I didn't do more testing to check robustness or temperature, etc. So, I have no idea what well happens on long runs at high speed. If you have a more proper setup, then such tests are possible and interesting to know the behavior.

FirePT, 2020-08-06:
@benjaf 
> @FirePT If you need more speed without using excessive amounts of power, you might consider modifying the motor to a 'delta' configuration (see issue #26). This trades some of the torque for speed, which seems like a good deal in this case.

I'm currently reading about the delta configuration and I may try it thank you. Just saw this [twitter post](https://www.google.pt/url?sa=t&rct=j&q=&esrc=s&source=video&cd=&cad=rja&uact=8&ved=2ahUKEwiXl7m-5obrAhU2aRUIHXwUDM0QtwIwAHoECAIQAQ&url=https%3A%2F%2Ftwitter.com%2Fjanhenrikh%2Fstatus%2F1129767318018580485%3Flang%3Den&usg=AOvVaw0rsCMnd7OGXBFriq2tqWds) and got excited with the possibilities :smile:

> @FirePT
> 
> > do you think it's safe to increase this value in terms of integrity of the motors/board ?
> 
> You can increase that number to 1500 to avoid fixed point overflows. Personally, the maximum I went was around 1200 rpm. It was for a short period, but it felt quite dangerous, and I didn't do more testing to check robustness or temperature, etc. So, I have no idea what will happens on long runs at high speed. If you have a more proper setup, then such tests are possible and interesting to do.

Thank you. I'll continue testing and share the results I got.

closed 2020-09-18 by EmanuelFeru

Candas1, 2020-10-22:
Just one thought, have you used the maximum speed coefficient?

FirePT, 2020-10-23:
Hi @Candas1 I don't believe so... What do you mean by that? How does it map to settings file ? 
I've been away since our last talks and wasn't able to further advance my project, that's the reason for all my silence, I think that I will have some free time on during my Christmas vacations.

Candas1, 2020-10-23:
This is the default speed coefficient:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/4fbf2549c979eeebf393ff067568cf7e58425570/Inc/config.h#L181

The maximum value would be 32767 ( Higher value would be negative ).
This is also influencing final speed ( if no other bottleneck I guess ).

I am wondering if you would gain even more speed.

kris86k, 2022-09-01:
Hi.
Any newses? I need ~3000RPM for small grass cutter.

Candas1, 2022-09-01:
3000rpm is not possible with this firmware

Candas1, 2022-09-01:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits#o-speed-limit

kris86k, 2022-09-02:
Can you give me the optimal settings for the max RPM? 
It's hard for me to understand.

Candas1, 2022-09-02:
What is the point you don't understand ?