EFeru/hoverboard-firmware-hack-FOC #87: How high voltage can the motherboard handle?

odox, 2020-08-09:
I have an electric moped with a bldc hub motor (1500 W) that I modified from led acid batteries to a bunch of 18650 li-ions a year ago. I had to change the bldc controller to one that I could be  programmed. But this controller makes rumbling noise at low speeds that I don't want. Since I use your brilliant FOC controler in my robot lawnmower I also want FOC control to my moped. But the battery pack is 14s and is almost 58V when fully charged. I guess is a bit to much for a hoverboard mainboard. Has any one tried to modify the board to handle higher voltage and current? Is it possible?

juodumas, 2020-08-09:
I'm driving with a 16S2P LiFePo4 battery pack, which I charge to 58.4V. I haven't seen any issues with the higher voltage, but LiFePo4 drops voltage very rapdily to 54V. 14S Li-ion sounds more risky.

Another issue that you might have is 1500W/56V = 26A for a single channel. This much current definitely works for brief periods, but the board definitely wasn't designed for this much current. I guess you will have to try and see if it burns :)

gerkon56, 2020-08-12:
I am using 67 v li-ion.
On a 2wd scooter. I also replaced my native dc-dc with dc-dc from aliexpress. The native dc-dc 67-14v transistor heats up strongly.

odox, 2020-08-13:
Thanks for your answers! I might take the risk of trying ;-)
Do you also charge your batteries via the motherboard?

gerkon56, 2020-08-14:
No i charge from ana

closed 2020-09-18 by EmanuelFeru

gutabsgf, 2023-01-17:
> 我用的是67 v锂离子电池。骑着两轮摩托车。我还把我的原生dc-dc换成了速卖通的dc-dc。原生dc-dc 67-14v晶体管发热强烈。
使用啥型号的DC-DC