EFeru/hoverboard-firmware-hack-FOC #88: Limit speed for backward movement

Gdenich, 2020-08-10:
Dear Emanuel,

I have finished the tricycle project based on your firmware for my children. I use the HOVERCAR variant and use the DOUBLE tap on brake pedal (button) for backward movement mode. Is it possible to limit the speed in backward movement mode, so that no one could enter full throttle by mistake when moving backwards.

Regards,
Denis

EmanuelFeru, 2020-08-10:
By default is not backward limit is not in the software at the moment. However it shouldn't be difficult to add. In main.c, you just need to limit the `speed` variable, for example  

`speed = MIN(speed, your_limit)`  

If, `MultipleTapBrake.b_multipleTap` variable is True.

closed 2020-09-18 by EmanuelFeru

SergiBori, 2021-01-11:
> Дорогой Эмануэль,
> 
> Я закончил проект трехколесного велосипеда на основе вашей прошивки для своих детей. Я использую вариант HOVERCAR и использую ДВОЙНОЕ нажатие на педаль (кнопку) тормоза для режима движения назад. Можно ли ограничить скорость в режиме движения назад, чтобы никто не мог по ошибке войти в полный газ при движении назад.
> 
> С уважением,
> Денис

> Дорогой Эмануэль,
> 
> Я закончил проект трехколесного велосипеда на основе вашей прошивки для своих детей. Я использую вариант HOVERCAR и использую ДВОЙНОЕ нажатие на педаль (кнопку) тормоза для режима движения назад. Можно ли ограничить скорость в режиме движения назад, чтобы никто не мог по ошибке войти в полный газ при движении назад.
> 
> С уважением,
> Денис

Привет. Можешь более подробно рассказать, как уменьшить скорость заднего хода. Сделай, пожалуйста, скриншоты.